bitdump/BLHeli #569: Need version 32.7 for SpeedyBee_4in1_V2

KarlKokosboll, 2021-11-25:
My new SpeedyBee ESC is running 32.8, but it randomly switches motor direction and sometimes some propellers won't spin or will only spin periodically. This sometimes fixes itself by just unplugging the battery several times but that gets very tedious and unreliable after a while. Is it possible for me to get the older 32.7 hex files which have a stable version like I saw someone get for a FOXEER REAPER ESC?

sskaug, 2021-11-26:
We just found a bug in the code that is dedicated for the MM32SPIN160 MCU that is used in your product. This bug causes the code to hang in some occasions. We are working with SpeedyBee to verify a fix, and once it is fully verified we will release it to the market.

KarlKokosboll, 2021-11-26:
Hello! Thank you for replying so quickly. 

I see, thank you for working on a fix. But in the mean while, would it be possible for me to use 32.7, while waiting for the fix to be verified? If 32.7 hex files exist for this ESC.

Thank you again!

sskaug, 2021-11-27:
This has nothing to do with 32.7 or 32.8. This MM32SPIN160 MCU was not supported in 32.7. It requires quite a few lines of specialized code, and it was in this new dedicated code for MM32SPIN160 that a line reenabling an interrupt was missing. Tescode fixing this is now posted here for the SpeedyBee MCUs:
https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/MM32SPIN160%20testcode

KarlKokosboll, 2021-11-27:
Ah alright, thank you very much! I'll try that and let you know if it works!

KarlKokosboll, 2021-11-27:
I have tested it and everything seems to work fine now! Thank you very much Sskaug!

closed 2021-11-27 by KarlKokosboll

lowefpv, 2021-12-01:
hi ive been having the same issue and have had 3 replacement escs the same thing on each esc 2 corrupted itself issues ran from smoked motor just sputtering and death roll i was sent 32.82 from speedybee claming it was supposed to fix the issue and same thing happened i came to github and saw the beta 32.83 read the notes and hoped it would fix after testing on pack 7 it corrupted again mind you this was a longer run than the previous 2 escs and i was thinking it was fixed blheli confirmed corrupted esc seemed the firmware on all others were 32.83 and on that one said corrupted and 32.8  really hoping this can get fixed soon as speedybee seem to be more towards replacing instead of refund and i just want to fly not have this issue with new stacks i just bought