bitdump/BLHeli #570: Speedybee ESC "found firmware revision 36.82"

Bigboostfpv, 2021-11-28:
[image]
 what can i do?

4712, 2021-11-28:
@Bigboostfpv This is a result of a memory corruption. We are working to implement a solution for this and will get back soon with a beta version including a fix, which enables a reflash. Is it the windows version you need?

Bigboostfpv, 2021-11-28:
Yes.
SpeedyBee_4in1_V21_Multi_32_82.Hex ?
thanks

Von: 4712
Gesendet: Sonntag, 28. November 2021 12:42
An: bitdump/BLHeli
Cc: Bigboostfpv; Mention
Betreff: Re: [bitdump/BLHeli] Speedybee ESC "found firmware revision 36.82"(Issue #570)

@Bigboostfpv This is a result of a memory corruption. We are working to implement a solution for this and will get back soon with a beta version including a fix, which enables a reflash. Is it the windows version you need?

4712, 2021-11-28:
@Bigboostfpv Please check out[ this beta exe file](https://app.box.com/s/druoa2bt1bfkaq6du93mjjucdmcsj1y1). Simply replace it with the original 32.8.1.2 exe.

Bigboostfpv, 2021-11-28:
How to flash ESC #3?

Von: 4712
Gesendet: Sonntag, 28. November 2021 20:56
An: bitdump/BLHeli
Cc: Bigboostfpv; Mention
Betreff: Re: [bitdump/BLHeli] Speedybee ESC "found firmware revision 36.82"(Issue #570)

@Bigboostfpv Please check out this beta exe file. Simply replace it with the original 32.8.1.2 exe.

Bigboostfpv, 2021-11-28:
[image]
[image]
[image]

4712, 2021-11-28:
I hope this shows what to do, should be quite easy...

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Bigboostfpv, 2021-11-29:
After first flash and reboot, motor 3. won't spin.
After second flash, it jumped back to 32.8??? flash was verified correctly?
[image]

After 3. flash
[image]
it works so far.

But there is a second problem with INAV. The dshot beacon (an beeper) is always on after power up.
"beeper -all" didn't help
But i guess this is a INAV problem, bf works fine.