bitdump/BLHeli #572: High PWM frequency - can't raise throttle

blind-oracle, 2021-11-30:
ESC: Tekko32 F4 65A Metal
Motors: T-Motor F90 1300kv
BLheli_32: 32.8

The configurator allows me to set PWM up to 128khz on this ESC.
If I'm using e.g. 48-128khz variable PWM it works fine.

But if I'm setting it to e.g. 96khz fixed or 128khz fixed then the motors are armed ok, spinning at idle speed, but raising the throttle has no effect at all - they continue spinning as is.

64khz fixed works fine, maybe a bit more, haven't tried.

What can be a problem here? Thanks in advance

sskaug, 2021-12-01:
This is strange. Can you share a screenshot of BLHeliSuite32 after you have read setup?

blind-oracle, 2021-12-01:
@sskaug Sure, will do tomorrow. There's nothing special there, most settings are on the default.

Meanwhile i've found (using 8khz increments) that 72khz is probably the highest stable frequency. On 80khz throttle works, but RPM seems very unstable and quad rotates over yaw axis badly even on the ground while trying to take off.

sskaug, 2021-12-02:
Ok, I see. This makes sense. Rampup power sets a limit on max pwm for low rpms. And when lowering it, you lower the allowed pwm for a given rpm. And with higher pwm frequency, you can get a lower rpm for a given pwm.

closed 2021-12-02 by sskaug

blind-oracle, 2021-12-03:
@sskaug Thanks.

I've tried today flying with the following settings set to non-default:
- PWM: 128khz
- Sine Modulation: On
- Motor timing: Auto

(this is a long range build -> trying to squeeze some % of efficiency)

And had persistent drone drops (probably) due to ESC rebooting - after a drop I hear a startup tone from motors.
It just randomly crashed 2 times in 5 minutes of flying. What can be the cause? Or I shouldn't play with 128khz? :)

krunked-fpv, 2021-12-03:
You should be using 32.8.3 48khz low, by RPM high PWM. 23 timing.

blind-oracle, 2021-12-06:
@sskaug Downgraded to 96khz and flew fine for quite a long time - and then at home during testing ESC resetted again. Fortunately it was very low above floor.

I think there's some kind of a bug here...