bitdump/BLHeli #573: SpeedyBee 4in1 V2 ESC

emty1106, 2021-12-02:
Hi, 
I wanted to ask if the bespoken "disarm when beeper gets turned on"-problems with bidirectional DShot on the Speedybee ESC are strictly Software based?
Or could it be a hardware problem?

SpeedyBee themselves just told me to not use bidirectional DShot until it got fixed, but.. if the only problem is the beeper, then I could just deactivate it right?
Except there is a hardware defect,

gomaed, 2021-12-02:
Same for me, I'm really wondering what is going on. Could we please get a feedback whats going on. Right now I don't trust this ESC and it would be nice to get some more technical response what is going on.

I have a quite new V2 ESC/FC Combo and I also have seen that they don't seem to use the "standard" MCU+Fortior MOSFET driver combo. It's only one IC on the daughter board were all seems integrated and the markings have been scratched off. On an older video from Joshua were he reviews the ESC I have seen they used a different daughter/controller board with seperate MCU+Drivers. Has it something to do with this new HW maybe?

Best Regards

emty1106, 2021-12-03:
> Has it something to do with this new HW maybe?

Yeah I would think so!
This guy shows the issue pretty well - https://www.youtube.com/watch?v=QdhkFb8I_Ms&t=1s

But the newly released software which is supposed to fix the problem with that MCU has a new bug.
Reports say, their motor smoked after a short flight with that new testcode.

gomaed, 2021-12-20:
Hello again,

so I was brave enough to test the newest version 32.8.3 (not 32.8.2). I have done several flights from very gentle to quite hard ripping. For me everything work right now. 

- Music: OK
- DSHOT beeper: OK
- Protocol: DSHOT600
- Bidirectional DSHOT: OK
- RPM filter: OK
- Battery: 4S (no additional Capacitor)

Not glitches or other issues on my side anymore right now.

Best Regards

CrucioFPV, 2022-01-02:
> Hello again,
> 
> so I was brave enough to test the newest version 32.8.3 (not 32.8.2). I have done several flights from very gentle to quite hard ripping. For me everything seems to work right now.
> 
>     * Music: OK
> 
>     * DSHOT beeper: OK
> 
>     * Protocol: DSHOT600
> 
>     * Bidirectional DSHOT: OK
> 
>     * RPM filter: OK
> 
>     * Battery: 4S (no additional Capacitor)
> 
> 
> Not glitches or other issues on my side anymore right now.
> 
> Best Regards

I tried version 32.8.3 too and everything seems to work fine

richschreuder, 2022-02-15:
> Hi,
>
> I wanted to ask if the bespoken "disarm when beeper gets turned on"-problems with bidirectional DShot on the Speedybee ESC are strictly Software based?
>
> Or could it be a hardware problem? [...]

I'm also having problems with the same ESC. Speedybee reached out to me and asked if I wanted to test the beta firmware of blheli. It seems to resolve the problem however I've got a new issue with one ESC getting invalid after a couple of flights.