bitdump/BLHeli #575: Substantial start in opposite direction

neoxic, 2021-12-13:
First off, thank you for this amazing piece of firmware!

I mainly use **BlHeli_S** ESCs as hydraulic pump and track motor drivers in construction vehicle models, i.e. in a non-standard way. These are **SPEDIX** 25A and 30A ESCs with targets **Q-H-15 MULTI** and **Q-H-20 MULTI** respectively. Having played a lot with different configuration settings, I am still facing one annoying bug. A motor briefly jerks in the opposite direction then corrects itself. It happens both in normal and bidirectional mode. I am aware that when the firmware is just starting, it proceeds blindly without prior knowledge of the rotor's position, etc. But... I don't see such an issue with the proprietary ESCs that I have. I mean, maybe it's there, but it's negligible. Yet with **BlHeli_S**, the time a motor sometimes spends in brief reverse is quite substantial. For example, a model's track visually moves backwards for a while taking into account that it's not 1:1 - there's a multi geared transmission at play. In other words, there needs to be considerable amount of reverse rotation to produce a profound movement.

I played a lot with the timing and startup power settings. No substantial difference. The issue is more or less omnipresent. I tend to believe that more "throttle" produces more of it rather than less, but I can observe it with low "throttle" as well.

Could changing targets from MULTI to something else help in resolving this issue? Any other suggestions?

I could also take a deeper look into the code myself, but I'd appreciate some guidance as to where to start looking and whether there's some technical document that describes the current algorithm or something. Thank you kindly!

sskaug, 2021-12-13:
Starting up a sensorless brushless motor is not trivial. And in your case they also probably have a significant static loading. BLHeli_S code comes with the comments in the code as guiding, but not much beyond that. And coding in assembly is tedious.
For BLHeli_32 we have made significant improvements, and also issued a code that is improved for situations with static loading: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Loaded%20startup%20testcode

neoxic, 2021-12-13:
There's almost no load in my case as most applications are heavily geared. And I tested both with load and no load and couldn't see any correlation. I understand very well that starting a sensorless brushless motor is not a trivial task. It's just weird to see some Chinese ESCs with which the motors just go. I just press and they go! The problem with them that make them barely usable is that they tend to have a pretty high minimum non-zero throttle level (roughly around 30-40%). That  means that a motor can't be started gradually without dashing, and that's a nuisance for me. In terms of diagnosing the issue, I tried to emulate abrupt transition from 0% to 30% (using another MCU for example) for BlHeli_S to no avail.

Hmm... I've always thought BlHeli_32 is closed source.😊 I guess l'll take a look into it then. But again, if static load is not the case, would it still make sense to switch from BlHeli_S, or is there's otherwise no significant difference in terms of algorithmic approach? I'm asking because I guess I'll need to replace a number of ESCs then.

sskaug, 2021-12-13:
There are significant changes in BLHeli_32. Just try out one as a start.

neoxic, 2021-12-13:
Oh, I'd better give it a try then. Thank you kindly for all your help and advice!

closed 2021-12-13 by neoxic