bitdump/BLHeli #576: Version 32.82 not working for SpeedyBee 4in1 V2 ESC

KarlKokosboll, 2021-12-13:
Hello!

A few weeks ago I made a post about version 32.8 of BLHeliSuite32 not working for the SpeedyBee 4in1 V2 ESC, the motors wouldn't spin or would randomly switch directions or ESCs wouldn't load in the firmware, etcetera. This was due to some glitch I learned. After about a day I was given testcode, what ever that might mean, for version 32.82 which I was told would solve the problem. I tried it and I said that it worked, but quite quickly afterwards I realised that nothing had changed, the exact same issues were appearing. I messaged SpeedyBee about this as well and they told me to stop using version 32.82. I am now back on 32.8 and the exact same issues are still happening. I seem to remember reading it was due to some line of code causing the ESC to freeze or hang, maybe it was called MCU or something similar, I am not sure. In any case, I need help. Have the issues been fixed yet or are they being fixed, if so, might you know in how long they will be fixed?

Thank you very much for your kind help once again!
 - KarlKokosboll

sskaug, 2021-12-13:
Please try out the 32.8.3 code: https://github.com/bitdump/BLHeli/releases
It has a fix for the MM32SPIN160 MCU used in your ESC.

KarlKokosboll, 2021-12-13:
Hello. Thank you for your quick response! But I have heard that the 32.8.3 code causes the motors to permanently break in some way, smoking the motors I believe. Was that also fixed now or is that still a risk? I don't want to accidentally break a motor.

Thank you once again!

KarlKokosboll, 2021-12-17:
@sskaug Hey again, I'm wondering if you got the chance to read my question above yet?

sskaug, 2021-12-18:
There has only been published one Rev32.8.3, which is now installed in hundreds of systems without any serious reports of issues.

KarlKokosboll, 2021-12-18:
Very well, I'll try it and see if it solves the issue! Thank you for your help :)

closed 2021-12-18 by KarlKokosboll