bitdump/BLHeli #578: Telemetry data noise from Airbot Wraith32 ST (Blheli_32, 32.8.2-32.8.3 firmwares)

DJ-URAN, 2021-12-19:
Hi all! 
Temperature, current and voltage indicators behave very strange when I get telemetry data from my Airbot Wraith32 ST ESC.
ERPM data are obtained more or less accurate and I can normally control the motor.
I made test measurement data of telemetry with a survey frequency of 20 Hz (every 50 ms) by DSHOT150 protocol and received data on the UART to a computer. As you can see, the temperature (orange line)  readings are unstable, right up to zero. Also, these voltage (green line) and current (grey line) are constantly changing, especially sharply when the temperature data is zero. Why does this happen and can it be connected with the firmware? I tried it with the latest firmware 32.8.2 and test 32.8.3, but did not get a difference.
[image]

sskaug, 2021-12-20:
The way telemetry was initially implemented, was from a request in a dshot frame. And then the end of telemetry transmission was also detected and handled in the input signal input interrupt routine. And this is also the way end of transmission is currently done. So if doing auto telemetry (every 32ms), you must have an RC signal rate higher than that. Or if requesting telemetry from a short input signal (<30us), you must have a new input signal pulse without a request before issuing a new request. Maybe that is the problem here?

DJ-URAN, 2021-12-20:
> 

No!
I send dshot150 signal for every 1ms (1kHz), and ask esc for telemetry every 10ms (set telemetry bit in dshot packet).
I also try get telemetry every 50-500ms. Situation with temperature was same

sskaug, 2021-12-21:
Are you sure your telemetry input is configured with pullup (or no pullup or down)?

DJ-URAN, 2021-12-21:
> Are you sure your telemetry input is configured with pullup (or no pullup or down)?

Shure with pull up!
It is not matter in my issue, cause i get telemetry data in first post by USB-UART (CH341A 3.3V level) adapter! Not by my pcb board!

DJ-URAN, 2021-12-26:
Below is telemetry data from ESC with throttle =48. The time between telemetry request is 10 ms.
I think, firmware have troubles:
[image]