bitdump/BLHeli #582: Motor Error rate while using Dshot 600 and Rpm filtering

Bob-lance, 2022-01-05:
Hi everyone I just built my first drone got it on the bench and am noticing around 2-3% error rate while spinning the motors up in betaflight.  I got no error when I switched over to Dshot 300 and everything seemed to be fine.  I would really like to be using Dshot 600 and do not no what I am doing wrong to get these errors.

Esc: Diatone Mamba 50a bl heli 32
Fc: Diatone Mamba mk2 f7
motors: Fpv Cycle 25mm 1880kv

gattimattia, 2022-01-11:
When you get an error rate higher than 0 is probably because cpu load reaches 100%; You can decrease cpu usage by disabling accelerometer and barometer and turning off data logging. There are probably other things you can do to lower it.
If none of these works you should switch to 4k dshot300, probably you won't notice any difference between 8k dshot600 but it's all subjective.
Hope it helps!

simoccc, 2022-01-23:
Hi, I have near the same issue , I don't know if I must open another separate issue case.

Brand new FC and ESC.

FC = Matek F722 HD
ESC = iFlight SucceX 50a BL32
Motors = Brotherhobby 2806.5 1300kv

Betaflight 4.2.11

BLheli32 :

32.7  = DShot 300 no issues - DShot 600 Errors ( Low errors from 0.02 sometimes 0.2 ) 

32.8 = DShot300 no issue - DShot 600 No issues but motor 4 RPM telemetry data is oscillating while the others seems good ( In Betaflight configurator motors tab and in BLheli32 suite motor tab )

Do you have any recommendation? Can I trust 32.8 with this motor4 data ?

haslinghuis, 2022-01-23:
Would have to test as not all hardware is the same. I would compare 32.8.3 and 32.7.x