bitdump/BLHeli #585: Motor timing > 31?

limonspb, 2022-01-23:
Hello!

Is that possible to make a test hex build that allows motor timing to be set > 31 or hardcoded to higher values?
I am working on a little project, about how to make a freedom spec drone faster, doing some measurements.

Freedom spec is a quadcopter 5'' 6S 2000kv motor, but you put a 3S battery on it, so it really struggles with speed and acceleration.
The efficiency is not a problem, and i can see the motor timing 31 seems to work really good, gaining 1%+ of the top RPM.

So i would be very curious to try motor timing values 31+, like 35, 40, 45.

Running FOXEER_Reaper_Mini_4IN1_45A_Multi esc for my test.

Thanks!

[image]

sskaug, 2022-01-23:
It is not possible to go beyond 31 (which is actually 30 degrees).
At 31, commutation is done immediately after a zero crossing is detected.

limonspb, 2022-01-23:
> It is not possible to go beyond 31 (which is actually 30 degrees). At 31, commutation is done immediately after a zero crossing is detected.

thanks for the explanations!

closed 2022-01-23 by limonspb