bitdump/BLHeli #586: 32.83 - motor RPM readings comparison

limonspb, 2022-01-24:
Hello guys.

I am testing the same ESC/same setup/same settings, different ESC firmware

On the left - 32.7, on the right - 32.83.
If that's not a secret, how did you manage to achieve that level of noise in 32.83 test code?
Is it some internal filtering, and if so, what is the delay on it in comparison with 32.7?
(32.8 showed more noise than even 32.7, but i guess you know that)

Thanks!

[image]

sskaug, 2022-01-24:
You can find some info on the algorithms used here: https://github.com/bitdump/BLHeli/issues/579
And the reason 32.8 and higher has a slightly different rpm reading, is that the software timing (regarding handling of the hardware timer) is made more accurate.

limonspb, 2022-01-24:
> You can find some info on the algorithms used here: #579 And the reason 32.8 and higher has a slightly different rpm reading, is that the software timing (regarding handling of the hardware timer) is made more accurate.

ok thanks! so its not the motors going slower, it's just readings are more accurate?

sskaug, 2022-01-24:
I can't say 100% for sure, but I do think it is only that readings are more accurate.
The reduced noise is both due to more accurate software handling of the hardware and due to some filtering added,

limonspb, 2022-01-24:
> I can't say 100% for sure, but I do think it is only that readings are more accurate. The reduced noise is both due to more accurate software handling of the hardware and due to some filtering added,

I can see you mentioned "sliding average of weighing the new sample with 0.5".
Could you please clarify, is that a moving average with a certain window size, like 10 or 20 samples or a time-based window?
Or there is no window and it's something like:
`new = 0.5*new + 0.5*old` a-ka low pass filter?
?

sskaug, 2022-01-24:
This is the code:
      if (Rpm_Period < 512)
         Rpm_Period = ((3*Rpm_Period) + Cnt)/4;
      else
         Rpm_Period = (Rpm_Period + Cnt)/2;
"Cnt" is the new measured period, and "Rpm_Period" is the averaged value.

limonspb, 2022-01-24:
thanks for the clarifications! I am excited about 32.83 :)

closed 2022-01-24 by limonspb