bitdump/BLHeli #587: 32.83 PWM - byRPM

limonspb, 2022-01-25:
Hello. Sorry for spamming with the questions.

I am testing more 32.83 firmware and noticed it has a new option for max PWM: byRPM.

Could you please elaborate on how this one works?
It seems to work a little better than just max value, give a tiny bit more RPM, but could be just my measurements error.

Thanks!

sskaug, 2022-01-25:
There is a FB group, but not sure how active it is. It is not run by us devs.
Check out the test code description here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM
And some posts on RCG, like this one: https://www.rcgroups.com/forums/showpost.php?p=47968903&postcount=4552

limonspb, 2022-01-26:
@sskaug thank you!

closed 2022-01-26 by limonspb