bitdump/BLHeli #595: Support in ESC Layout AT vs ST for TEKKO32 F3 and F4

moserello, 2022-02-08:
Hello all!

We had an esc that burned mid flight and we replaced it with a brand new one… The new one is a Holybro Tekko32 F4 45A; the three originals esc are Holybro Tekko32 F3 35A.

We tried this new esc in BlHeli32 Suite and everything seemed fine but when we tested the motor linked to the new esc (in the motors tab) it wasn’t able to reach max throttle. It was fine until 8000 rpm, it raised its speed like the other motors, but then it just stopped at that value (not so high). If it goes eventually full throttle its value stays up like the other motors for some milliseconds and then it drops to 8000 rpm again. We noticed also that after few seconds on the bench without props on at 8000 rpm the motor was way too hot compared to the other 3 motors (at about 30000 rpm).

We tried to connect another functioning (brand new) motors to that esc and it gives the same result, so we can assume it’s an esc problem… motors work well with other escs

Looking at the tabs we noticed that the new esc differs from the others in different aspects… **It’s AT while the others are ST**, and it says **Bootloader l** while the others say **Bootloader j**.

Is this linked some way to this motor “weakness”? Is there something we can do about it? Is it possible maybe to flash a AT esc to make it a ST esc? (If that’s a possible solution…)

[Log_esc.zip](https://github.com/bitdump/BLHeli/files/8022868/Log_esc.zip)

[image]
[image]
[image]
[image]
[image]

sskaug, 2022-02-09:
No, it is not possible to change the code of the ESC.
But from what you describe, it sounds like it may be the temperature protection of your AT ESC that is the problem.
This ESC does not have a working temperature sensor, and temperature protection must be set to off.

moserello, 2022-02-09:
@sskaug thanks for your reply and help! A lot appreciated 

> No, it is not possible to change the code of the ESC.

Thanks, this one is a fixed point, that solves a huge question mark.

> This ESC does not have a working temperature sensor, and temperature protection must be set to off.

Also tried with temperature protection disabled (in this test I was trying to set the same parameters for the 4 escs and so that it is set as protected). 

The weird thing is that:

- I can setup the esc with an arduino nano COM interface, but I can't flash FW as the only one available is "32.8 (sv)". No way to see other versions. Maybe the target is wrong?
- If I connect the esc to the FC and I use blheli32 suite, the 4th motor isn't recognized, BUT i can spin it in the motor tab
- Tried swapping motors and escs on the same FC branch (#4) but all is working fine and I can exclude other failures
- Maybe there's a suitable version of blheli32 (e.g. 32.7) that I can download and try to flash?

[image]

Thanks!

4712, 2022-02-09:
- if an ESC was started after a previous revision was superseded, let's say after 32.8 was released, then there is no older revision available.
- The AT type ESCs usually need some more wait time for the bootloader to connect.
If using a FC as interface, just click connect first and the power up the ESCs if possible. 
Or click "Connect" , then wait 5 seconds and click "Check" afterwards.

4712, 2022-02-10:
The (sv) stands for loaded from remote, not yet stored local in the dll. 
As I understand the fw revisions are all 32.8?
May be the ESC is defective...

4712, 2022-02-10:
> Sorry for the dumb question, that means that the FW is on the ESC only? I'm interested in a deeper understanding if you don't mind.
The firmware is always only on the ESC. But if you want/need to flash it again or upgrade, the fw is loaded locally (dll) or from remote.

closed 2022-02-14 by moserello