bitdump/BLHeli #597: 1 of the ESC gets status ***INVALID***

richschreuder, 2022-02-14:
Hi, I'm testing Rev32.8.3 on my Speedybee 4 in 1 esc. This was asked to me by Speedybee itself cause the bidirectional shot issue on the latest hardware version of the 4 in 1.

For a couple (3) of times during a TestFlight the quadcopter crashed because 1 off the motors stopped spinning. This always happens on Esc number 4. Every time when I check the ESC with blheli suite the 4th ESC gets the status ***INVALID***. After reflashing the ESC everything's good for a while (5 to 10 battery packs) en the issue reveals it self again. 

Does anyone have the same issue? Is this issue related toe the beta firmware? Am I missing something else?..obviously I checked all the wires and soldering that seems all good..help is greatly appreciated!

sskaug, 2022-02-15:
We have no indication that the "invalid" issue is related to 32.8.3 code.
Please contact your vendor about your "invalid" case.

richschreuder, 2022-02-15:
> We have no indication that the "invalid" issue is related to 32.8.3 code.
> 
> Please contact your vendor about your "invalid" case.

Hi sskaug,

Thx for your reply!
That's what I did in the first place...the only thing I got back was a bigger question mark 😂

Grtz.

haslinghuis, 2022-02-15:
@richschreuder can you replicate with 32.7

richschreuder, 2022-02-16:
> @richschreuder can you replicate with 32.7

Hi, I did not try. This ESC has problems with 32.7 and bidirectional DShot.. it causes motors not to spin during arming..the first ESC blew up (the drone was on fire) so.. I'm not very in to this again..

Speedybee sent me a new one btw so kudos for that.

Bigboostfpv, 2022-02-16:
Hi, do you want the hex from issue #570. It works on my V21 for 20-25 flights?
[SpeedyBee_4in1_V21_Multi_32_83.zip](https://github.com/bitdump/BLHeli/files/8084129/SpeedyBee_4in1_V21_Multi_32_83.zip)
plz check if the file size is different.

richschreuder, 2022-02-16:
> Hi, do you want the hex from issue #570. It works on my V21 for 20-25 flights?
> 
> [SpeedyBee_4in1_V21_Multi_32_83.zip](https://github.com/bitdump/BLHeli/files/8084129/SpeedyBee_4in1_V21_Multi_32_83.zip)
> 
> plz check if the file size is different.

Hi! Thx, will check this tomorrow!

richschreuder, 2022-02-17:
> Hi, do you want the hex from issue #570. It works on my V21 for 20-25 flights?
> [SpeedyBee_4in1_V21_Multi_32_83.zip](https://github.com/bitdump/BLHeli/files/8084129/SpeedyBee_4in1_V21_Multi_32_83.zip)
> plz check if the file size is different.

Hi, i've checked the HEX file size...and yours is 1KB bigger than the one im using.

i'll flash my ESC with yours..see if it helps.

thx!

richschreuder, 2022-02-18:
> Hi, do you want the hex from issue #570. It works on my V21 for 20-25 flights?
> 
> [SpeedyBee_4in1_V21_Multi_32_83.zip](https://github.com/bitdump/BLHeli/files/8084129/SpeedyBee_4in1_V21_Multi_32_83.zip)
> 
> plz check if the file size is different.

Flew 7 Packs today without any problems!

Bigboostfpv, 2022-02-18:
Nice, plz don't use var PWM / by RPM