bitdump/BLHeli #599: Question: Tekko32 F3 4in1 65A Metal 96k PWM possible?

HazzaHFPV, 2022-02-16:
This is more a question than an issue. Would it be possible to run the Tekko32 F3 Metal 65A (I believe it is Revision A) also with up to 96kHz PWM? I know Rev. C is capable, I just can't get my mind around the fact that this F3 ESC cannot do it (when my small BlHeli_S ESCs can).
I own two of these:
https://www.3dxr.co.uk/multirotor-c3/multirotor-escs-c48/holybro-tekko32-f3-metal-4in1-esc-65a-replaced-by-31097-p3318

Like I said, not really an issue, more of a wish :) Maybe there is a way. Nonetheless, thanks to all the developers for their work on this and  making these ESCs possible.

sskaug, 2022-02-17:
This is not quite straightforward. The pwm frequency range is locked in the bootloader by the hardware vendor, to a range that they qualify their hardware for. 
Besides, the very high (generally beyond 48k) pwm frequency ranges are not providing any benefit in most systems. That is except for the very small motors (whoops) that have very low inductance. 
Max power does not at all depend upon pwm freq, as when driving the motor at full power, the pwm is 100% (which is the same as no pwm).

HazzaHFPV, 2022-02-18:
Thank you very much for this insight and taking the time to answer my question. I was brought onto the hype-train by a recent video from Joshua Bardwell that basically states that any ESC in the current days that doesn't go to and beyond 96k is rubbish, and that this would be the most important feature to look for when buying ESCs from now on.
https://youtu.be/pVmff9Id5X8

Your comment and insight is therefore most valuable to me, as I bought these ESCs believing they were "quite top notch" until I saw his video.

sskaug, 2022-02-18:
Yes, unfortunately a lot of info there is inaccurate and misleading. Which we have made Joshua aware of.
All MCUs (F0, F3, F4, G0 etc all!) support "byRPM".
All MCUs (F0,F3, F4, G0 etc again) support any pwm frequency range. The pwm range that an ESC is supporting is determined by tht hardware vendor from what their hardware design supports (not relating to MCU, but to caps, fets etc).
And then there is a significant marketing hype these days to go high in frequency.

For most systems, byRPM does not go very high in frequency: https://www.rcgroups.com/forums/showpost.php?p=47911559&postcount=4539
Here you can see a 2500kV motor on 4S, goes just a bit above 40kHz.

FritzWlan, 2022-03-01:
Is there maybe a chart showing 6s motors around 1900kv? I wonder if they also work best at 16-by rpm settings.

sskaug, 2022-03-01:
A 1900kv on 6S corresponds approximately to a 2850kV on 4S. So about 14% higher pwm frequencies than in that plot. Which is in this context pretty much equal.

closed 2022-06-04 by HazzaHFPV