bitdump/BLHeli #604: Flash testcode on mac?

limonspb, 2022-03-09:
Hello dear developers.
What is the process of flashing testcode on Mac?
On Windows you download hex file and put it into BLHeli32_HexFiles folder, then restart the app.

What about Mac? Where is that BLHeli32_HexFiles folder located?

Thanks!

4712, 2022-03-09:
It should be user/documents/BLHeliSuite32xm/BLHeli32_HexFiles or use the file select dialog (here under win, almost similar under macOS ) 
[image]

daleckystepan, 2022-03-10:
You need to pu Hex files into `iCloud Drive/Documents/BLHeliSuite32xm/BLHeli32_HexFiles`

closed 2022-03-11 by limonspb