bitdump/BLHeli #605: 32.7 write setup failed

IZIRC, 2022-03-21:
Enviment:
flycolor X cross BL 32 50A version 32.7
USB link programar Esc BlHeli_32
win10

Steps
1.connect interface(USB/COM)
2.read setup and then power the ESC.(read setup successful)
3.set ESC some values, click write setup.

Expected:
should be successful!

Actual:
Popup message:
1.
Warning
Write Setup section to ESC Failed!
This means: the ESC layout might no longer be identified automatically until next power cycle.
The layout information is:

Flycolor_X-Cross_BL_32 - Rev. 32.7 - Multi

Please analyze the reason of failure, cycle power and try to reflash the ESC.
[Log.zip](https://github.com/bitdump/BLHeli/files/8318101/Log.zip)

4712, 2022-03-21:
@IZIRC this, for sure, should not happen. I assume, you can still read the setup, but write always fails...
It looks like, if the ESC someone got write protected...

To sort out the interface as the reason for the issue:
What is "USB link programar Esc BlHeli_32" exactly, do you have a link? 
Did you try another interface, like Arduino Nano?

IZIRC, 2022-03-21:
[image]
[image]
 
https://youtu.be/UkXtWBAaHKw

Pincho USB para programar Esc BlHeli_32 bits:
https://a.aliexpress.com/_v4cnDb

xkillogrande, 2022-04-30:
Hello,

I have the same issue with the same ESC and same USB adapter....
I can connect and read the ESC but cant flash/write anything to the ESC.

I would love to have help with this issue.

4712, 2022-04-30:
@xkillogrande could you please provide a log file?
(Options... Enable Log, try a write, save the log to log.tlg and post here)

xkillogrande, 2022-04-30:
This is what I have on the log.
I can also try to make a video, if it helps.

[Log.zip](https://github.com/bitdump/BLHeli/files/8597799/Log.zip)

4712, 2022-05-01:
Did you try another interface, like Arduino Nano?

xkillogrande, 2022-05-02:
I followed this video [](https://www.youtube.com/watch?v=OXdwnjbbdiM) and tried with a Arduino Uno and Mega.

[image]

Both of them could not connect/read the esc and would get "stuck" on the print screen above.

xkillogrande, 2022-05-02:
The video:

https://www.youtube.com/watch?v=OXdwnjbbdiM

4712, 2022-05-02:
The Pin to connect the ESC depends on the hex file for the Arduino Uno you flash.
Which hex file did you choose and at which pins do you connect the ESC?