bitdump/BLHeli #606: Smoking Motors on infinity305 esc

big0013, 2022-03-23:
been flying Fettec G4 and f7 on 32.8.3 and all are flying so damn smooth, but miner prop strikes have cause multiple smoked motors, bells are fine and props are fine. Same tune as 32.7 and never had this issue with light taps.  I have smoked a total of 4 motors on three different quads using the infinity 305 esc. on 16- 48 by rpm. timing at 23, demag low , ramp up 50, any idea as to whats going on? I'm thinking the demag setting might need to be turned up. I believe they added another setting like very high or something, so maybe its not lowering power. I guess I'll go back to 32.7 again, but kinda bummed as 32.8.3 is feeling so good. P.S. motors are not getting hot on 32.8.3 during normal flight just light taps confuse the esc and motor burn up.

Kevinfpv, 2022-03-23:
I had the same experience yesterday. 2 drones with smoked motors after hitting the gate slowly. Also no hot motors during a test flight, but after hitting the gate lightly, the drone fell down with a burnt motor. With 1 drone i re-armed and my esc sets on fire. AIKON AK32PIN 35A 2-6S on 32.8,3  16-48 by rpm.

daleckystepan, 2022-03-23:
Can you please provide blackbox from flight with burnt motors?

Kevinfpv, 2022-03-23:
Sorry i was not logging my flights.

big0013, 2022-03-23:
I'm running kiss without a logger.

Kevinfpv, 2022-03-28:
today back to 32.7 with my new esc's . No problems when i crash or hit a gate :) Bye bye 32.8.3

big0013, 2022-03-28:
I switched to 24 to bye rpm and haven't had anymore issues.

Bubbows, 2022-04-18:
Yup, I just experienced similar on 32.9.0.3. Tried variable pwm by rpm feature, 16 - 48 and smoked 3 motors after a light tap on a gate. Something funky is going on. Unfortunately blackbox flash was full by the time the motors smoked but I can give you the logs prior to that if it helps at all
[220418_gts v3 working well but death by heli32 variable by rpm no good.zip](https://github.com/bitdump/BLHeli/files/8507361/220418_gts.v3.working.well.but.death.by.heli32.variable.by.rpm.no.good.zip)
.

Bubbows, 2022-04-18:
Sorry, got that wrong, pwm setting was: 16 - variable by rpm. (my esc is a reaper 60A mini v1 maxes out at 48khhz pwm I think)

Slinkywinky, 2022-06-02:
I've has similar experiences with TMotor Velox V2. Smoked all four motors after bouncing off of several branches in a tree. going back to 3.17. This is the 3rd quad I've burned motors on using this firmware. Why is no one talking about this?