bitdump/BLHeli #607: 32.89 and Foxeer ReaperMini 60A 128

limonspb, 2022-03-27:
Hello!

Wondering what's the difference between 32.83 and 32.89?

And could you please compile hexes of 32.83 and 32.89 for
Foxeer Reaper Mini4IN1 60A 128.
There are hex files for 96, but not for 182
[image]

Thanks!

sskaug, 2022-03-27:
The 32.8.9 testcode is only for applications that have some load upon startup, like underwater thrusters.
Rev32.9 is currently in the release process (it is equal to 32.8.3 code), and will be available for all targets in a few days.

limonspb, 2022-03-30:
@sskaug thanks!
what is the due date? :)

I have a competition coming this weekend, wanted to run the most recent firmware for it :)

sskaug, 2022-03-30:
It is coming in the very few next days.
But I slipped your code into this folder: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Rev32.8.3%20testcode

sskaug, 2022-03-30:
It is coming in the very few next days, hopefully today or tomorrow.

closed 2022-04-01 by limonspb