bitdump/BLHeli #609: BLHeli can't read ESC of SPEEDYBEE-BL32-45A-ESC-V22

Bi2Lmet, 2022-03-31:
Hi, I'm new in the club...
I'm trying to flash the SPEEDYBEE-BL3-V22 but unfortunately the BLHeliSuite32_32.9.0.1 (or any other older) can't read the ESC.
-I have flash the SpeedyBee F7 V2 with Betaflight firmware 4.3.0  and I can connect ok via USB... but I can't read the Setup or I can't flash, check or verify.

-Also even all the settings inside Betaflight have NO Arming Disable Flags, I CAN NOT make the motors to spin!  
Also got 0.00A in ESC censor at battery, but the battery it's ok, I can see it at the meter on the top.

Is it something that I need to do, or something to check?

4712, 2022-03-31:
If no ESC can be read, maybe it is about something wrong with the power system...
Do they play beeps on power up, and if, how many?

Bi2Lmet, 2022-04-04:
I've got some times 2 beeps and some other 3 beeps (not like the start fast 3 beeps, slower).
-Also I've noticed that the is a small part missing from ESC... so there is certainly a problem!
https://drive.google.com/file/d/1-v5gWS_LCtcihIbrU9_BCyB-bJJe8lXF/view

I believe I need a new ESC... or some how to find that missing part