bitdump/BLHeli #610: BLHeli 32 - latest release 32.09.02 - on Mac OSX - 10.15.3 (Catalina) crashes

mrRobot62, 2022-04-01:
After installation the latest BLHELI32 release on my Mac it crashes after starting. Window pops up for a second than I got below crash-report.

If needed I can attach details crash - report

`Process:               BLHeliSuite32xm [12368]
Path:                  /Applications/BLHeliSuite32xm.app/Contents/MacOS/BLHeliSuite32xm
Identifier:            BLHeliSuite32xm
Version:               1.0.0 (1.0.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           BLHeliSuite32xm [12368]
User ID:               501

Date/Time:             2022-04-01 11:55:28.047 +0200
OS Version:            Mac OS X 10.15.7 (19H1824)
Report Version:        12
Anonymous UUID:        2C580062-5D3A-989E-2624-EA9E21A12CBA

Sleep/Wake UUID:       A4BD3A3B-986E-4215-8B30-087C75E7AF06

Time Awake Since Boot: 1600 seconds
Time Since Wake:       400 seconds

System Integrity Protection: enabled

Crashed Thread:        12

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
/usr/lib/libcrypto.dylib
abort() called
Invalid dylib load. Clients should not load the unversioned libcrypto dylib as it does not have a stable ABI.

Thread 0:: Dispatch queue: com.apple.main-thread
`

mrRobot62, 2022-04-01:
need urgently 32.8 version - could you please provide this stable version - thanks

4712, 2022-04-01:
@mrRobot62 thanks for the info, will be fixed asap, meanwhile please find version 1.0.3.2 [here](https://drive.google.com/file/d/1VLcO0N49jir9GIMRuKVkI9fUSdZRuQRi/view?usp=sharing)

4712, 2022-04-02:
@mrRobot62 It is fixed now in BLHeliSuite32xm rev 1.0.3.8.

mrRobot62, 2022-04-02:
uh - ups - try to download 1.0.3.2 from google drive:
[image]

Error says: File is in Wastebox from owner :-O

1.0.3.8 is currently not uploaded as release right?

4712, 2022-04-02:
1.0.3.8 is to find here:
https://drive.google.com/drive/folders/1Y1bUMnRRolmMD_lezL0FYd3aMBrNzCig
or here:
https://www.mediafire.com/folder/dx6kfaasyo24l/BLHeliSuite
Github release is not yet updated.

mrRobot62, 2022-04-02:
[image]

Super - thanks - works as designed :-)

4712, 2022-04-02:
Great, Thank you!