bitdump/BLHeli #611: BLHeli 32 - latest release 32.09.02 No 32.9 file for Hobbywing 60a ESC

enrico320i, 2022-04-01:
Hello, i can not flash my Hobbywing 60a ESC with the Latest 32.9 it only shows as latest 32.8?
I used beta 32.8.3 before fine?

Greetings Enrico

sskaug, 2022-04-01:
Hi Enrico. Yes, that's the way it is. HW want to have a longer time for testing release code before publishing. 
HW 32.9 code will be published, but it may still take more than a month.

enrico320i, 2022-04-01:
Thanks for the info, so for now 32.8.3 latest. No problem because it works, was just wondering. Did a Quad Setup day and first startet with a T-Motor ESC and then the Hobbywing no 32.9.. :)

mituritsyn, 2022-04-10:
[image]
dalrc engine 40A also unavailable

4712, 2022-04-11:
@mituritsyn Now, you can give it a try...

lacasitoFPV, 2022-05-09:
Good night, any news regarding the release for the Hobbywing XRotor and the 32.9 version? Kind regards!

4712, 2022-05-09:
Just released, please retry.

digitalanthill, 2022-05-09:
Flashing now. Thanks for the update

lacasitoFPV, 2022-05-10:
> Just released, please retry.

Thank you very much! I will flash it tonight. Kind regards!