bitdump/BLHeli #612: Mamba 4 in 1 ESC blew up after 32.9 flash

Harald777, 2022-04-06:
Just flashed Blheli32 32.9, coming from 32.8 test code, and within 3 minutes 2 FET's from my Mamba 45A 4 in 1 ESC blew up. 
Could this be Blheli32 related??

[image]
[image]
?

sskaug, 2022-04-07:
Sorry to see your loss. By now we do have thousands of installs of 32.9 and the testcode equivalent of 32.8.3 with very few reports of problems. So most likely your case is unique and not relating to firmware.

Harald777, 2022-04-07:
Okay, pure bad luck.

Thanks fort he feedback.

 

Greetings,

 

Harald

dan0412, 2022-04-23:
Hello, I flashed 32.9 on my MAMBA F55_128K BL32 4IN1 ESC 55A for a new build. Upon arming and throttling up, the motors sputter and then reboot the FC. A quick google search took me to this link.

https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Betaflight%20bidir%20Dshot%20glitch%20issue

Is this a fix for the MAMBA F55_128K BL32 4IN1 ESC 55A? I want to be sure before I download the .hex file and flash. Or should I revert to 32.8?

Thanks