bitdump/BLHeli #613: Xilo_4in1_V2 - Persistent **INVALID** error with 1 motor

philboltt, 2022-04-08:
Hi,
I'm having a persistent issue using the XILO Stax 45A BLHeli_32 3-6s 4-in-1 ESC.  The issue manifests very similarly to that described in [issue 597](https://github.com/bitdump/BLHeli/issues/597). After flashing the firmware and setup the quad flys perfectly for a few batteries, but invariably errors again and when re-connected to BLHeli it reports an invalid motor (motor #1 in our case).  Re-flashing the firmware solves the issue for a few batteries, until it reappears.  At first we though it was a hardware issue, but we've now replaced both FC and ESC and the issue persists.  The behaviour is consistent with 
The firmware hex that we're using is the `xilo_4in1_v2_multi_32`.  We've tried with BLHeli_32.8 and 32.9, but there is no change to the issue between these versions.
I noticed that BLHeli 32.8 added the XILO_4in1_V2 hex file.  Forgive me if I'm getting my dates wrong, but it looks like this ESC was released in 2019, whereas BLHeli 32.8 was released in 2021.  Does the XILO_ESC_B hex work with this ESC?  It seems to be the version that was available prior to the 32.8 release.  I can't find anywhere online to download BLHeli 32.7 to check against  XILO_ESC_B, and I haven't had any luck trying to use 32.8 or 32.9 to flash the firmware with XILO_ESC_B.
Any advice or assistance in testing against 32.7 is greatly appreciated!

sskaug, 2022-04-08:
You can only load xilo_4in1_v2 code to your ESC. And for the type of MCU used in this product, earlier code revs than 32.8 are not available. Your issue may be hardware related, you should contact your vendor.

philboltt, 2022-04-08:
@sskaug Thanks for the response.  I've sent a message to the vendor.