bitdump/BLHeli #614: Half-PWM Firmware

FritzWlan, 2022-04-11:
Hi,

not a real issue, but more like a feature request. Is it possible to publish more half pwm firmware versions? There are several FCs not supporting PWM-Freq below 48Khz. Would it be a big hassle to publish half-pwm firmware for all these ESCs?

Thanks in advance!

Myllerman, 2022-04-14:
> There are hundreds of codes by now. And only a few would want half pwm freq testcode, so we would prefer still to have it on a request basis.

Could you update the Half pwm of the"AIKON_AK32PRO_50A_4IN1_MINI_6S_Multi" to the newest version?

sskaug, 2022-04-15:
Now posted

FritzWlan, 2022-04-18:
I would like to request the latest FW half_pwm for: Tekko32_F3_C_Multi_32_83 , thanks!

sskaug, 2022-04-18:
Now posted.