bitdump/BLHeli #617: 1.0.3.8 Linux exceptionally slow

nerdCopter, 2022-04-15:
1.0.3.8 exceptionally slow in Linux -- especially connection and disconnection.

`Linux 5.13.0-39-generic #44~20.04.1-Ubuntu SMP Thu Mar 24 16:43:35 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

4712, 2022-04-16:
@nerdCopter What FC and firmware version?

nerdCopter, 2022-04-16:
ApexF7/ESC stack (STM32F722/Furling32_4in1_C): https://pyrodrone.com/products/apex-f7-stack-w-60a-esc
EmuFlight 0.4.0
ELRS 2.4Ghz
Video of slowness: https://youtu.be/ODQiYhz6b4A

Have 3 quads. All EuFlight. Sometimes it was fast(normal) after repeat connections. Sometimes slow.
HelioSpring v1/HobbyWing_XRotor/EmuF 0.4.0/ELRS_900Mhz - was faster, but saw it slow once or twice.
HelioSpring AIO/Furling32_Multi/EmuF 0.4.0/ELRS_900Mhz seemed fast/normal always.

4712, 2022-04-16:
Looks really slow, weird. Will look into it asap.

4712, 2022-04-17:
@nerdCopter 
Can you provide a screenshot of the log as shown?

[image]

Also a cli "diff" in the EmuFlight configurator would be helpful.

nerdCopter, 2022-04-17:
Please inform me how to "log" in BLHeli, in order to provide your request.

[EMUF_cli_nerdAPEX6_20220410_140127_preFlight_dABG700from800_throttle0.30from0.40.txt](https://github.com/bitdump/BLHeli/files/8502170/EMUF_cli_nerdAPEX6_20220410_140127_preFlight_dABG700from800_throttle0.30from0.40.txt)

4712, 2022-04-17:
Do "options.. enable Log", do a connect and switch to "log" tab.

nerdCopter, 2022-04-17:
some fast, some slowish, but not as bad as video.
4 screencaps:
Fast
[image]

Slowish
[image]

Fast
[image]

Slow
[image]

4712, 2022-04-17:
@nerdCopter thanks for all the useful records. Will need some time to reconstruct what happens here.

howels, 2022-04-18:
Seeing same slowness in Linux on both 1037 and 1038 - both of the 32.9 releases so far. Ubuntu system, connects to a wide range of hardware fine but cannot read SkyStars Kramman 55A on 32.9.0.2 or 32.9.03 (blhelisuite 1.0.3.7 and 1.0.3.8)
Works fine on 32.8 and can flash the 32.8.3 test code ok.

packocrayons, 2022-04-18:
Also seeing this issue in 1038 - BLHeliSuite32 is pinning a CPU, doesn't seem to be doing any I/O though.

4712, 2022-04-18:
Now I'm able to reproduce.
The issue seems related to the system. Using a VM, the problem does not occur, but using real hardware, it does...

packocrayons, 2022-04-18:
What's changed between releases? Does the blheli_s configurator get updated at the same time? It's open source and if the bug followed it will be easier to help debug

4712, 2022-04-19:
> Now I'm able to reproduce. The issue seems related to the system. Using a VM, the problem does not occur, but using real hardware, it does...

Sorry, I was too hasty. I got confused by having more default usb/serial ports (ttyACM0..2) and tried to connect ttyACM0 while the FC was at ttyACM3...  so, issue not reproduced yet. Will do further investigations.

4712, 2022-04-19:
The change was after 32.8.1.2/1.0.3.2. I guess something around heartbeat-timer usage, which changed, or the switch to a newer IDE version. I will need a little time to generate some test versions and get back here.

4712, 2022-04-19:
Here a link to [version 1.0.3.2](https://app.box.com/s/e69xid4jtpi5rn9r12e0qdwqlzv5z56v)

packocrayons, 2022-04-19:
4 intel i5 M520 cores. On an IBM thinkpad T510. Nothing exceptional by any means

nerdCopter, 2022-04-20:
will test/troubleshoot 1.0.3.2 this weekend.
cpu:
```
$ lscpu 
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   39 bits physical, 48 bits virtual
CPU(s):                          12
On-line CPU(s) list:             0-11
Thread(s) per core:              2
Core(s) per socket:              6
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           165
Model name:                      Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz
Stepping:                        2
CPU MHz:                         900.070
CPU max MHz:                     5000.0000
CPU min MHz:                     800.0000
BogoMIPS:                        5199.98
Virtualization:                  VT-x
L1d cache:                       192 KiB
L1i cache:                       192 KiB
L2 cache:                        1.5 MiB
L3 cache:                        12 MiB
NUMA node0 CPU(s):               0-11
Vulnerability Itlb multihit:     KVM: Mitigation: VMX disabled
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Enhanced IBRS, IBPB conditional, RSB filling
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_t
                                 sc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm
                                  pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb st
                                 ibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsa
                                 veopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp pku ospke md_clear flush_l1d arch_capabilities

```

4712, 2022-04-20:
Thanks for the input!
Here are some test versions: 
[beta1](https://app.box.com/s/7hahap628khtblyaf9xji6cjofhau0cc) same as 1.0.3.8, but compiled with ide of version 1.0.3.2
[beta2](https://app.box.com/s/e6ls4nkqyuug2tl696zmk2m7x93jelro) same as 1.0.3.8, but w/o high frequent checking port availability.
[beta3](https://app.box.com/s/br3netat384rrsf3uayvpmp8uky5ikkg) same as 1.0.3.8, but w/o timed heartbeat.
Beta2 and 3 do have log enabled and will save a log file on close of app in the app folder.
If it does not work better, please post the log.xlg files as *.txt here. (The log files can now be loaded into the log tab)

howels, 2022-04-21:
> Thanks for the input! Here are some test versions: [beta1](https://app.box.com/s/7hahap628khtblyaf9xji6cjofhau0cc) same as 1.0.3.8, but compiled with ide of version 1.0.3.2 [...]

Those 3 versions all exhibited the same bug, when reading the ESC it would appear to timeout but show the firmware version when I tried to close the window?

Beta2 -
[BLHeliSuite32xl-beta2.zip](https://github.com/bitdump/BLHeli/files/8535189/BLHeliSuite32xl-beta2.zip)

Beta3 -
[BLHeliSuite32xl-beta3.zip](https://github.com/bitdump/BLHeli/files/8535171/BLHeliSuite32xl-beta3.zip)

nerdCopter, 2022-04-21:
same conclusion.
beta1, 2, 3 all exhibit same behavior but possibly differing levels of slowness as before.  did not really repeat/retry (i.e. not deep troubleshooting)
[beta3_220421_164827_log.xlg.zip](https://github.com/bitdump/BLHeli/files/8535334/beta3_220421_164827_log.xlg.zip)

beta3 slow but tolerable
[beta2_220421_165330_log.xlg.zip](https://github.com/bitdump/BLHeli/files/8535335/beta2_220421_165330_log.xlg.zip)

beta2 excruciatingly slow.
beta1 slow, then fast, then slow, then fast again :P

build 1037 slow. noticed "checking for updates" stayed on statusbar.
build 1032 flawless, but outdated and not supporting newer firmware.

4712, 2022-04-21:
@howels & @nerdCopter thank you for testing, still hope to find, whats going on...
Could you please do another short test with "Options... expert Mode"/"Options... Use Legacy Progress Dialogs"
[image]
?

howels, 2022-04-21:
Hangs when attempting to read ESC config, log files are generated in the program directory when I try to close the unresponsive program, initial 445 byte file is only thing created before I try to close the program.

[BLHeliSuite32xl-beta2.zip](https://github.com/bitdump/BLHeli/files/8535709/BLHeliSuite32xl-beta2.zip)
[BLHeliSuite32xl-beta3.zip](https://github.com/bitdump/BLHeli/files/8535711/BLHeliSuite32xl-beta3.zip)

nerdCopter, 2022-04-23:
beta 3 expert legacy dialogs **fast(normal)**: [beta3_legacy_220423_081634_log.xlg.zip](https://github.com/bitdump/BLHeli/files/8546989/beta3_legacy_220423_081634_log.xlg.zip)
 
 beta3 expert legacy **slowish**, but not excruciating: [beta3_legacy_220423_081915_log.xlg.zip](https://github.com/bitdump/BLHeli/files/8547001/beta3_legacy_220423_081915_log.xlg.zip)
 
 edited a prior comment above.
 observation: sometimes clock options or other menu's will not open menu, needing re-clicking. maybe an overall UI issue is causing this as possibly suspected?

howels, 2022-04-23:
After opening beta3 again, it is faster but still not a normal speed.

4712, 2022-04-24:
Thank you both again for all the testing. @nerdCopter Yes, it looks like that issue is the GUI update handling.
(opening menus are also known to be slow, will see, if we do here anything).
What about normal version 1.0.3.8 with legacy dialogs enabled, would it be usable?
Or is beta3 + legacy dialogs enabled required?

One part of the increased slowness is also a different connect and reconnect handling.
@howels where do you see the beta3 slowness most? Does it help to disable the log function and to set the Extended wait time on FC Connect to 500 ?

howels, 2022-04-28:
I've upgraded to Ubuntu 22.04 now, with new GTK versions and the app is even worse. Previously when I clicked "Port" to select the /dev/ttyACM0 interface I would get the menu appearing on a different monitor to the program itself, now the menu doesn't even show up.
I can't drag these newer versions either, any movement makes the window go beserk and flicker. 1.0.3.2 I can at least drag around.
The developers need to test this on Linux. I am using BLHELI-S, JESCm BlueJay and AM32 also and blheli32 is the only one with problems under Linux.

howels, 2022-04-28:
I cannot clear the log option. Clicking it once it is enabled causes the UI to freeze and I have to force close the app.Lowering the timeout from 5000 to 500 makes no difference.
Essentially unusable, I do not trust this to not crash during flashing.

nerdCopter, 2022-04-28:
I've not had time to troubleshoot further yet.  I can concur that in general, 3rd party GTK3 apps are a problem. I've noticed many projects offering GTK2 and GTK3 releases as separate downloads, fwiw, and I'm forced to choose the GTK2 versions for any usability.  Sorry, i know this is not a "forum" but felt it pertinent information for the conversation..

4712, 2022-04-29:
In about a week, I will have better test environment. 
Thanks, for the input about GTK2/3 and also ubuntu 22.04.
Some of the UI move/resize issues are due to the compatibility with Elementary OS 5.x, which most likely can be dismissed after 6.x was released.

nerdCopter, 2022-04-30:
* the released 1.0.8.3 with Legacy Dialogs enabled is still slow.
* the trick to the menus, is to wait for it to be highlighted by cursor-hover.
* beta 3: the initial connect/read is slow, but subsequent connect/read without unplugging is nominal/fast.
* beta3 logs attached (Legacy Dialogs): [beta3_logs.zip](https://github.com/bitdump/BLHeli/files/8597628/beta3_logs.zip)

4712, 2022-04-30:
Thanks again for the testing. Meanwhile I have set up one hyper-v with ubuntu 20.04.4 and one with ubuntu 22.04.
on both, there is no slowness of 1.0.3.8 (legacy dialogs or not), if the app window is totally in the visible area. If the statusbar is not visible it gets real slow.
I will do a next beta asap, which shows execution times for the connection steps in the log.

howels, 2022-04-30:
> Thanks again for the testing. Meanwhile I have set up one hyper-v with ubuntu 20.04.4 and one with ubuntu 22.04. on both, there is no slowness of 1.0.3.8 (legacy dialogs or not), if the app window [...]

Suspect that VMs might not be 100% representative of hardware. Using a 3 monitor setup here.

nerdCopter, 2022-06-04:
i was finally able to reproduce that weird window glitch.  it was simply by selecting "options"
https://youtu.be/Gu7I_Zy25CY

`Ubuntu 20.04.4 LTS`
`XFCE 4.14`
`Linux 5.13.0-44-generic #49~20.04.1-Ubuntu SMP Wed May 18 18:44:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

4712, 2022-06-05:
@nerdCopter Thanks again for the detailed reports. I tried with real machines ubuntu 20.04 and 22.04 gnome desktop and hyper-v ubuntu 20.04. gnome and xfce4, not sure which version exactly. I could not reproduce it ... 
I'm not about the references, but the menu seems buggy somehow...
 
Does this [beta4 ](https://app.box.com/s/psrivbxa8gojwkqh18ji066jsaxa48vc)change anything?
Introduces GTK3 window direct updating and skips window resize event handling.

howels, 2022-06-06:
> @nerdCopter Thanks again for the detailed reports. I tried with real machines ubuntu 20.04 and 22.04 gnome desktop and hyper-v ubuntu 20.04. gnome and xfce4, not sure which version exactly. I could [...]

Beta4 is back to normal speed! Working fast again 😄

nerdCopter, 2022-06-11:
> Does this [beta4 ](https://app.box.com/s/psrivbxa8gojwkqh18ji066jsaxa48vc)change anything? Introduces GTK3 window direct updating and skips window resize event handling.

beta4 worked good and speedy! :+1: :))  
i had enabled logs, but no files saved upon exit.
(test only included connect/read/disconnect. tested 3 quads)

4712, 2022-06-12:
@howels 
> BTW I copied the library and other files from beta3 as only the executable file was included.
that's fine, only the executable changed. 

@nerdCopter 
> beta4 worked good and speedy! 👍 :)) 

Great, then the next release will include the changes
> i had enabled logs, but no files saved upon exit. (test only included connect/read/disconnect. tested 3 quads)

The beta4 does not include the automatically turn on the log and save. The log file can be saved manually using the right mouse button menu.
Maybe it is useful to ask for saving the log on application closing, or add an auto save option...

howels, 2022-06-23:
Any timeline for the next release? BTW I still see the drop-down menus appear on different monitors, probably a old GUI toolkit issue.

4712, 2022-06-23:
@howels I'm still looking to solve the drop down menu issue, hope to get it done in some days.

howels, 2022-07-26:
Ping @4712 - checking on the status of this as I come to install BLheliSuite32 on a new machine.

4712, 2022-07-27:
@howels As it looks now: the fix for the menu items dropping down on the wrong monitor, if running on a secondary monitor, is still not fixed. ATM I tend to release a fixed version, with the known issue, that the app must run on the primary monitor. 
But I'm still checking to use another menu component.

4712, 2022-07-27:
@howels suddenly, after your motivation  ;-), I got the right idea, where to look in the SDK code, and it seems to be a fix possible.
Will get back with a beta soon...

nerdCopter, 2022-07-27:
While i understand the firmware will always remain closed-source, potentially the UI could be open-sourced for 3rd party to solve as well -- just a thought.

i "packaged" the beta 4 here in case someone needs in the interim: 
https://gist.github.com/nerdCopter/53b3495a683df2bdbc18ea332c513aec

howels, 2022-07-28:
thanks @nerdCopter. I've been downloading the old version and the beta4 each time. 

The blheli32suite release process is sub-optimal also, it's just a series of links to other hosting services.

I strongly concur that the client should be open source and freely distributed, even if some core libraries and the hex files are closed-source.

4712, 2022-07-28:
here is the[ Beta 5 ](https://app.box.com/s/tjsrdsjlqe26ok7nypx7xqks067tl72n)as complete zip package.

4712, 2022-08-12:
@nerdCopter , @howels now the [release version](https://app.box.com/s/xvdb34e2kq1zd6vg4d4q4730k881tz9d) is out, which will hopefully work better

closed 2023-03-20 by nerdCopter