bitdump/BLHeli #619: BLHeliSuite32xm rev 1039 not working on MacBook.

ClassicShooter, 2022-04-24:
Currently I have MacBooks on Monterey and Big Sur. 

On Monterey, when I click on the BLHeliSuite32XM (BX) icon it flickers for a second and then nothing else. 

On Big Sur, when I click on the BLHeliSuite32XM (BX) icon, I am prompted to go to my Security & Privacy settings and click "Allow Apps downloaded from:" and I select the "Open Anyway" option. The icon then pops up for a second on the bottom bar, and then goes away. 

Nothing ever opens up for either one of the MacBooks.

4712, 2022-04-24:
@ClassicShooter What happens if you start  BLHeliSuite32xm from the terminal?

ClassicShooter, 2022-04-24:
> @ClassicShooter What happens if you start BLHeliSuite32xm from the terminal?

Same issue. 

For my "temp" fix I did the virtual Microsoft 10 thing and then used BLHELI though that.