bitdump/BLHeli #622: Requesting procedure to change motor direction

David-OConnor, 2022-05-09:
Hi! I'm writing custom firmware, and am having a struggle setting motor direction (Or perhaps any command? The LEDs and beepers aren't working either).

Normal throttle commands work well. Due to this, and a logic analyzer, I think I'm sending the correct signals.

Procedure I've been trying (including variants of):

- Set throttle on all 4 motors to idle, using command *48*. (It appears command 0: *motor stop*, is not implemented?
- Wait ~1ms
- Send command *7* or *8* 6 times in a row, with 1ms after each, to set direction
- Send command *12* 6 times in a row to save, with 35ms after each.

What could be causing this? Thank you.

sskaug, 2022-05-09:
Do you also set the telemetry bit in these command frames?

David-OConnor, 2022-05-09:
Yes - I've been sending it in all frames.