bitdump/BLHeli #625: Desync on Foxeer Reaper 65A

Benxw, 2022-05-31:
Hello,

I just made my new build 5"

- ESC foxeer reaper 65A
- FC Hobbywing F4 G3 on Falco X
- Rcinpower wasp major 2020kv

I have problem with desync.
I tried to change motor timing or demag without succes.

My Foxeer reaper was coming with firmware 32.9
Maybe, can I downgrade to the 32.7 version.

How can I do that ?

Thanks for the anwers

sskaug, 2022-05-31:
Not sure exactly which code you have, but hopefully it is one of these: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

Benxw, 2022-05-31:
Thank you.
On blheli32, the name of my esc is : FOXEER_Reaper4IN1_F4_65A_128
one of the two codes above is compatible?

sskaug, 2022-06-01:
No, the name must be an exact match.
F4 MCUs were added to BLHeli_32 at the Rev32.8 level, and unfortunately are not supported for earlier code revs.

Tdogb, 2022-06-06:
I have this issue and it turned out to be one of the escs rebooting mid flight. Happened a 60A reaper too...then it happened to one of my buddies.

Tdogb, 2022-06-06:
Can we please the 32.7 binaries? My team would run those until blheli becomes stable again

sskaug, 2022-06-07:
Cases where the ESCs reboots are most likely related to hardware and not firmware. Besides, 32.7 is not available for F4 MCUs.