bitdump/BLHeli #629: PWM Frequency issue for Rush Blade 60A Super 96k

qsffffff, 2022-06-11:
Hey, I have an issue with the Min and Max PWM frequency, this esc is marked as 24 - 96 khz but i can only set it to 48 - 96khz. I want to set it in 24khz mode but i can't. Here is the official manual : https://ae01.alicdn.com/kf/Sbe23f54128244be896e5d30c214392f65.jpg?width=1520&height=6885&hash=8405

sskaug, 2022-06-13:
You can find testcode here:
https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Half%20pwm%20low%20frequency%20testcode
It will not allow you to set lower than 48kHz in the Suite, but when the Suite says 48k, the frequency is actually 24k.

kafie1980, 2022-06-19:
I came here to request this but my wish was already granted :)

Thank you for the testcode.

Already flashed the ESC's and tested motors on the bench.