bitdump/BLHeli #630: Holybro Tekko32 F4 45A - Temp telemetry reports 0 degrees C

0crap, 2022-06-19:
On my Holybro [Tekko32 F4 45A](http://www.holybro.com/product/tekko32-f4-45a-esc/) BLHeli32 ESC, temperature telemetry is not working.
It always reports 0 degrees C.
RPM telemetry is working fine.
I tested this with v32.8 and v32.9 with the same outcome.

This ESC is used in combination with a flight controller running iNav.
I have no clue if this is an issue with iNav or BLHeli_32.
I also reported the issue on the iNav Github as https://github.com/iNavFlight/inav/issues/8133

Thx for your time.

sskaug, 2022-06-20:
The F4 MCU that is used in this ESC has an internal temperature sensor that is not working properly.
So for some products, ESC manufacturers have chosen not so support temperature protection and temperature telemetry.
Which is the case for your ESC.
The way around this, is to use an external NTC resistor for temperature sensing, which some F4 products do.

0crap, 2022-06-20:
Ahhh well, thx for explaining that.
That's why the Temperature Protection is also greyed out.
Bit lame from the manufacturer to disable it instead of fixing it.
Anyway, that's why it's always reporting 0 degrees C.

Thx!

pdxtigre, 2022-06-20:
Not surprised to me. Had reliability issues with Tekko single ESCs as well. 

Holybro Tekko32 F3 35A 3-6s BLHeli32 ESC

Dsynced and smoked on my quad. 
The copper pads are very thin and got ripped on first impact. 
No temperature sensing. 

I have never purchased anything from Tekko ever since.  There are better choices in the market.

shellixyz, 2023-07-02:
> The F4 MCU that is used in this ESC has an internal temperature sensor that is not working properly. So for some products, ESC manufacturers have chosen not so support temperature protection and [...]

What do you mean by not working properly ? I haven't seen errata about the temperature sensor in these chips.

The main reason I'm wiring telemetry from the ESC to FC on my planes is to get the temperature. This situation is really annoying. I bought these ESCs expecting them after having checked that the MCU did have a temperature sensor to have temperature sensing like any other BLH32 ESC I have tested so far.

The worst is that it is still possible to set a temperature limit in the BLH32 suite which makes the ESC not work properly. It just limits the power all the time.

MrD-RC, 2023-11-20:
I've just seen that the 32.10 release has added the temperature, and other, sensor to bidirectional DSHOT. Are they also accessible over the telemetry pad?

0crap, 2023-11-20:
> I've just seen that the 32.10 release has added the temperature, and other, sensor to bidirectional DSHOT. Are they also accessible over the telemetry pad?

Thanks for the heads up.
I tested this and it works!
Putting a hair dryer on the ESC and the temp creeps up.
Bench tested with INAV 7.0.0 RC2

Well done!