bitdump/BLHeli #631: SKYSTARS SL40A Butter 128K: loaded startup testcode request

neoxic, 2022-06-21:
Hello,
Please publish loaded startup testcode for SKYSTARS SL40A Butter 128K ESCs.
Thank you!

neoxic, 2022-06-21:
As a side note, I can't wrap my head around Rampup Power being more than 100%, i.e. the maximum value of 150%. Could you please explain a little how it is possible to have 150%?

I'm trying to make a SKYSTARS SL40A Butter 128K ESC start a mini hydraulic pump motor quickly, i.e. from a complete stop to medium/full throttle. In most cases, setting Rampup Power to 100% worked very well. But recently, I stumbled across this one weird case when some sort of a startup delay is introduced even with Rampup Power 100%. It's not a hiccup, just a weird ramping delay before the motor goes mid/full throttle. Very annoying! Yet, setting Rampup Power to the maximum value of 150% greatly reduces the delay but doesn't eliminate it completely, so some additional info on Rampup Power could shed some more light on the behaviour. I'll also play with timing and with a loaded testcode when it gets published in hopes the delay goes away, but if there's anything else I can do, your advice would be greatly appreciated. Thanks!

sskaug, 2022-06-21:
Code is now posted. In this case Rev32.9.9, a sub-version of Rev32.9.
Rampup power is just a relative number - it just compares to a level of max power vs rpm that was at some point defined to be 100%.