bitdump/BLHeli #632: Tekko32 F4 45A ESC connecting BLHeli Software

tfischer90, 2022-06-21:
Hi BLHeli Commuinity,
I have an issue connecting with my Holybro Tekko32 F4 45A ESC. They are supposed to have BLHeli flashed. I dont know which version. I use a DYS USB Linker V2 and connect over the PWM and Ground. The ESC works with connected motor but it is in a strange mode where it slowly ramps up power to the PWM setting. I used the BLHeliSuite32_32.7.13 worked with the Holybro Tekko32 35A ESC which I used before. I treied to connect with the BLHeliSuite32_32.9.0.3 but I did not get it to connect. The ESC also does not support programming with sticks.
With the BLHeliSuite_32.7.13 I get an error that the ESC version is to new. With the BLHeliSuite32_32.9.0.3 the app stops working with the "connecting interface please connect ESC and power up".
Thanks a lot for your support in advance!
Best,
Tim

4712, 2022-06-21:
@tfischer90: Please check that the right com port is selected.
Try to connect and power up the ESC, wait 5 seconds and then click "Read Setup".

tfischer90, 2022-06-23:
Hi, we solved the problem with the usage of the BLHeliSuiteXL running on Linux was able to connect with the ESC. We used same Hardware just with Linux and that worked.
Thanks a lot for your help.
Best,
Tim

StuweFPV, 2022-07-27:
I have the same issue... but I don't have the opportunity to solve it with a linux version. Whenever i connect with 32.8 it tells me that this software will not support 32.9 but that's the build he detected on the Holybro. When running 32.9.0.3 i get the error "no valid config found" and "multiple esc detected" - basically nothing. so i'm stuck completely!
urgently need help to config my esc so i can go flying!!!

and why is it so f*** impossible to download older versions? I can't get 32.9.0.2 anywhere and on github it seems that there is only 1 version... what's going on??? i had some success in the past using slightly oder versions to get access. where did they go??? thanks to anybody that can help!

4712, 2022-07-27:
@StuweFPV I am sorry that you are experiencing such difficulties.
We could offer version 32.9.0.2 for download, but the only change in 32.9.0.3 was a fix in the server connection (32.9.0.2 will not be able to flash ESCs).

To understand the problem, it would be important to know exactly which interface (which flight controller? version of the installed firmware version?) you are using.

StuweFPV, 2022-07-27:
@4712 Thank you i appreciate your effort and help. I don't really care about flashing unless this brings the solution. I just in a first step need to be able to configure the ESC. Maybe 32.9.0.0 would do the trick? I have no idea which version Hollybro put on... seriously why can't i download old versions??

I'm using a Matek H743-Wing with iNAV 5.0 on it. I've made all sorts of changes to the protocol. Dshot 150, 300, etc. currently I think am on Oneshot125. thanks

4712, 2022-07-27:
@StuweFPV unfortunately several changes were done in the step from 32.8.x.x to 32.9.x.x...
Usually the newest version is the best to use, and often is mandatory because of important fixes.
Anyway here is [32.9.0.0](https://app.box.com/s/hfqfq9dmrheuokmjozc8fkmcohxif9p7) and [32.9.0.2](https://app.box.com/s/0vh89rok4pugh1mx0i7y843b3u94k6k0) for you.

I will look into Matek H743-Wing / iNAV 5.0 combination. (I have only a F765 here). 
It maybe also very helpful to provide a log file (log.tlg).
Enable Log (Options... Enable Log), do a check and then in the log window (Log... save to file...) and post here as zip.

4712, 2022-07-27:
@StuweFPV So far no issues here to see with MATEKSYS F765 WING and INAV5.0.

StuweFPV, 2022-07-28:
@4712 thank you for the files. i will try them out and see if it makes a difference. I don't fully agree that the latest is always the best. about 50% of my bl_heli ESC's i have to use older versions and can't update. some work others don't. i found this to be one of the most unreliable software i use. sorry to say that but it's honest. once i know which version works i never change it or update. the recent ESC's were just purchased last week so i was forced to try with 32.9 and almost "as usual" i run into big issues.
I did have the logging enabled and could not make any sense "general error" i think was the response. i'll log and provide at my next go. thanks for your help!

StuweFPV, 2022-07-28:
> 

maybe it's more on the Holybro side..? people seem to have heaps of problems with them. I own quite a few of those and the old ones were really good. this is my first set of v4 - maybe that's the problem...

StuweFPV, 2022-07-28:
no luck with any version...  here is the log file
[BL_Heli 32.9.0.3_Log.tlg.zip](https://github.com/bitdump/BLHeli/files/9213004/BL_Heli.32.9.0.3_Log.tlg.zip)
:

4712, 2022-07-28:
You're right, the log shows only that the bootloader of the ESC is not connectable.
But we don't give up on it ...
Now there are some questions to be answered.
- Does the ESC react to throttle input, if not, does beep on powering up, and how many beeps?
- You wrote that BLHeliSuite32 32.8.x.x showed a message that the app cannot handle firmware 32.9, can you please repeat this check under the exactly same conditions and make here a log file too?
-  Is it a single [Tekko32 F4 45A](http://www.holybro.com/product/tekko32-f4-45a-esc/) ESC?

StuweFPV, 2022-07-28:
- The ESC works fine when addressing it with inav. beeps propperly when powering up, led short flash and spins the motor how it should.
- i will try - the strange thing is that it takes about 20 reboots until i get 1 successful connection.. if i remember correctly that was when i lowered the speed to 9600bauds. but i can't reproduce it at will. it's more like a lottery win when and if he conencts
- yes correct model and yes only a single ESC connected
thanks for your help!

StuweFPV, 2022-07-30:
In my misery I remembered this old Omnibus FC that broke but i used him to config bl_heli ESC's as long as they were not properly connected to a FC or if I had issues. I connected my problem case (Holybro 45) est voila. works like a charm. I think i run inav 4.1 on it. so it looks very much to me that it has something to do with the FC and the release. I made a log file. hope it helps. thank
[BLHeli 32.9.0.3 Omnibus Log.tlg.zip](https://github.com/bitdump/BLHeli/files/9224197/BLHeli.32.9.0.3.Omnibus.Log.tlg.zip)
s

4712, 2022-07-30:
Perhaps an issue of the H743 input function at the motor out pin? Maybe try another pin or install Betaflight and check if bidirectional DShot works.
Another option would be to check the iNav config for anything disturbing the input function.

StuweFPV, 2022-08-01:
Hmm i'm not fully convinced of that. Once the ESC is programmed it works just fine. It's getting access in the first place is very very often the issue. So this is not my first case and i doupt that all these FC's have broken pin's for configuration but then they work fine for flying. Sorry to say but this seems very much a issue BL_Heli software not being able going through inav setup FC's to access the esc. I kinda gave up on it as my workaround with the Omnibus worked. Once it's all set I stick to my old wisdom of BL_Helis: never change a running system and NEVER EVER try to update

4712, 2022-08-01:
With iNav, the Motor pin is used always as output, never as input... unless it is used to get data from the ESC in order to establish a connection with BLHeliSuite32. I'm quite sure, that it is not an issue of BLHeliSuite32 software or BLHeli_32 firmware, not in this case. (BTW: If we never would have changed running systems, there would be neither BLHeli/_s/_32, nor BF/iNav and so on...)  
There are enough reasons thinkable, which can disturb the data flow, it's a pity, that we cannot go to the ground here.
Anyway, it's great, that it is working now for you!

StuweFPV, 2022-08-01:
I agree it does not make sense... but then again not many things do with bl_heli. I have had issues with that software for years. really a lucky guess if it works or not. somehow i always found a way around as all my planes fly with bl_heli but i must admit out of all the soft i use (inav, ET, pitlab, ardu, crossfire, etc.) bl_heli is the once by distance that gives me the most issues and headache. like i said, once it's configured I dare to never touch it again and i can fly. just a shame that it's like that... but it's free (at least it used to be) so beggars can't be choosers :) thanks for your help anyway

4712, 2022-08-12:
@StuweFPV There is a new version out, 32.9.0.4, which may work better with iNav F7 boards with single motor output..

IAMHAADICOOL, 2024-11-13:
Hey, @4712 I am facing the same issue, what should I do?

IAMHAADICOOL, 2024-11-13:
It's not showing in BLHeli32 suite, both in Linux and in Windows