bitdump/BLHeli #636: Curious Question on Assembler

snapo, 2022-07-01:
As i cant program in assembler.... How much faster is this assembler codebase compared to C?
C i would be able to read, is there a fork that uses C/C++ ? Out of interest...
There are also more people that write c/c++ than assembler.... 

This is not a issue / nor a feature request, it is out of curiousity. What i personally guess is a 10% or less speed improvement, but assembler is used because of the old Atmega chips forked from blheli with limited space 90%. If my assumption should be wrong.

Thanks for a answer

stylesuxx, 2022-09-10:
*Not the project owner, still trying to answer.*

The point is, when you write Assembler, there is no ambiguity, what you write is what you get. With C you will get what the compiler "thinks" is best. there are some tricks you might be able to apply "manually" that the compiler simply does not know about.

I think it is difficult to quantify the improvements without actually writing it in C and comparing the results.

In fact over in the [Bluejay](https://github.com/bird-sanctuary/bluejay) project (which is a fork of BLHeli_S) we are planning on porting it to C via SDCC (this is a long term goal though). If you are interested you can join us on Discord.

snapo, 2022-09-11:
Correct C optimisations from the compiler can be much slower / and or take more space.
Regarding Bluejay, i will monitor it, but i am not that good in development that i can take part of it :-) i am just able to hack some arduino code together but dont even get interrupts yet.

The main reason for asking this question was to be able to change the code and re-flash the quad ESC controllers and maybe use those quad ESC controllers in a way to control 3 gimbals (assuming the ESC has current meassurement via a motor controller or similar), those quad ESC controllers are very cheap compared to the FOC boards. So thats why i did want to tinker with it. Also the possibility to remove all remote control like pwm, and so on and set the parameters for example via i2c .... it just opens up some more posibilitys. With Assembler i am completely lost (yes a shame....)

stylesuxx, 2022-09-11:
I personally don't believe in re/purposing ESCs. They are simply so purpose built that making different things out of them is just not worth the hassle (unless you want to use them as switches obviously), especially since you can get ESP32s boards for just a couple of Dollars.

Timers are very important and most of the stuff in ESC firmware is timer based. I can highly recommend reading the reference manuals for the chips you are interested in. And learning Assembler is time well spent in my opinion. You just learn so much about how hardware works that it improves your overall understanding (also you learn to appreciate high level programming languages).

That being said, diving straight into ESC firmware is quite the feat. If you are playing around with Arduino I would recommend you to look into Assembler produced from your current projects.