bitdump/BLHeli #639: 32.7/32.83 firmware for foxeer reaper 60a 128k

Hrisfpv, 2022-07-16:
Is it possible to get one of these firmware because I’ve seen some bad things happening on 32.9.

sskaug, 2022-07-16:
Maybe, possibly. There are many versions of Foxeer Reaper out there.
What is the exact name of your firmware, as displayed in BLHeliSuite32?

Hrisfpv, 2022-07-16:
FOXEER_ReaperM4IN1_F4_60A_128_Multi_32_9

sskaug, 2022-07-16:
The MCU used on your ESC is not supported in 32.7 code.
But 32.8.3 is now published here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

svt913, 2024-06-17:
Hi, it seems the link to the firmware is no longer available. Will you be able to upload the file for the FOXEER_ReaperM4IN1_F4_45A_128 Rev 32.9 firmware? I have problems with the rotation of one motor in flight of a racing drone