bitdump/BLHeli #641: With temperature protection turned on my quad can only hover a couple of centimeters above the ground

sanj3k, 2022-07-24:
Hi, I am using two FOXEER Reaper F4 128K 65A BL32 4in1 ESC in a shendrones thicc X8 copter. Yesterday I couldn't maiden it because it was unflyable after a lot of debugging I miraculously decided to turn off the temp protection and the quad has all the power which was lacking before. 

Out of curiosity I tried to Debug Blackbox Log the esc temperature from the ESC telemetry but in the osd it said 8degC and in the BBlog 0degC so I am not sure about that. On top of that since I run two 4in1s I cant switch the esc telemetry wire to the other one without a lot of work in case it would be the lower one having maybe a temperature sensor problem (?)

Unfortunately there is only 32.9 available to flash. Any idea if maybe there is something messed up in the hex ?

sskaug, 2022-07-25:
Unfortunately these ArteryTek F4 MCUs have a temperature sensor that is not usable. The best way to work around this, is to add an external temperature sensitive resistor (NTC) for temperature sensing and protection. Howerver these have not yet been included in all ESC designs. So the way FW works for these non NTC equipped ArteryTek MCUs, is that temperature sensing always reports 0degC. And temperature protection kicks in if enabled - which as in your case can be confusing, but also is a way of telling users that temperature sensing is not supported. And the default setting for all these units is temperature protection disabled.

sanj3k, 2022-07-25:
Hi, first of all thank you for taking the time and explaining the situation. Should I then close the issues since it's known? 

Out of curiosity where should I solder this resistor in a 4in1 esc. But I think it's not going to happen to be honest :)

sanj3k, 2022-07-25:
Understood and yes at least someone now can google and find this thread. Once again thanks a lot

closed 2022-07-25 by sanj3k