bitdump/BLHeli #645: 32.8.3 firware for Foxeer Reaper F4 45 amp ESCs

BikeGuy1992, 2022-08-29:
I saw issue #639 asking for firmware 32.7/32.8.3 on the F4 60 amp Reaper and know 32.7 doesn't work with the F4 chips. 

Can we get a 32.8.3 firmware for the F4 45 amp Reaper ESC as well? I'm getting issues with all of mine where after barely clipping a gate one motor will just drop out on me and refuses to re-arm. Have tried doing an FC reboot but it doesn't do anything. I have to physically unplug and replug the battery back in for the ESC to reset. Other thing too is it isn't always the same motor that refuses to re-arm after barely clipping a gate. 

I've tried it on stock BLheli32 32.9 settings and my custom settings (35% rampup, demag high, motor timing auto, low PWM 24kHz, high PWM ByRPM) and it doesn't make a difference. Also tried putting FC settings (BF 4.2.9 and 4.3.1) back to stock as well and it doesn't matter. Would like to try it on 32.8.3 to see if that helps or not. When I asked Foxeer directly about it they put the blame on me saying their ESCs don't do that at all. I'm not the only one experiencing it. Me and a few other racers I fly with all have either the 45 or 60 amp Reaper, all of them F4 versions and they all do it. Appreciate any help.

sskaug, 2022-08-29:
Thanks for the good description. First of all, when an ESC goes unresponsive (with input signal and power present), that is something we have occasionally observed with some ESC designs. The MCU can go into a stuck state (generally termed HardFault) upon excessive noise in the system. We have no knowledge that that is what is happening in your case, just referencing some general experiences we have.

A gate clip will generally cause very high voltages and currents in the system, as the motor usually will stall and lose sync. We have looked at lowering these stall stresses, and can provide a testcode if you want. Probably not a dramatic change, but also most likely a reduction of system stress. Please let us know the exact name of your code as it appears in BLHeliSuite32 if you want to test.

Have you tried adding decoupling cap(s) to the battery terminal. A good cap can reduce voltage stresses.

As for 32.8.3, that was the testcode that was released as 32.9 - so they are identical (except for the rev no).

BikeGuy1992, 2022-08-29:
Thanks for the response. In terms of capacitors, I'm just using the included one that comes with the ESCs, a Rubycon ZLH 35v 560uF low ESR cap. 

I have some Rubycon ZLH 35v 1000uF I could try putting on them to see if that makes a difference or not. Where I'm putting the caps is directly on the ESC main pads since you want the capacitor hooked directly up to where the noise is being generated the most. The further away you put it from the ESC or where the noise is being generated, the less effective the cap will be. 

Is this something that could be recorded in black box when the motor does lose sync when one clips a gate? Not sure what debug mode I'd need to select in BF for it to show up. But figured maybe it might help you guys out in terms of trying to figure out what's going on with it. 

The Foxeer Reaper F4 45 amp name that is present in BLheli32 is...

FOXEER_ReaperM4INI_F4_45A_128 Rev 32.9

sskaug, 2022-08-29:
Now added 32.9.1 testcode here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes
Also added the 32.8.3 code for completeness.

BikeGuy1992, 2022-08-29:
Awesome, thanks for the help! 

Is is possible to add 32.9.1 testcode for the Foxeer Reaper F4 60 amp ESC as well? All of my race rigs use either the F4 45 or the F4 60 amp version of the Reaper. A few of them are the older F3 96K version (not the F383 128K equipped one, the version before that)  but I haven't noticed if those are doing the HardFault or not. Haven't been really paying attention. 

If you need the name for the 60 amp, it shows up in BLheli32 as...

FOXEER_ReaperM4IN1_F4_60A_128

svt913, 2024-06-17:
Hi, it seems the link to the firmware is no longer available. Will you be able to upload the file for the FOXEER_ReaperM4IN1_F4_45A_128 Rev 32.9 firmware? I have problems with the rotation of one motor in flight of a racing drone

BikeGuy1992, 2024-06-17:
> Hi, it seems the link to the firmware is no longer available. Will you be able to upload the file for the FOXEER_ReaperM4IN1_F4_45A_128 Rev 32.9 firmware? I have problems with the rotation of one motor in flight of a racing drone

Yeah, no firmware is available because BLHeli32 screwed everyone in the FPV community and took their servers offline over a week ago...

Whatever firmware you have flashed to it it is stuck with that firmware forever until the ESC dies in terms of BLHeli32 goes. 

Your best bet is to flash either AM32 or Escape32, or get a 8bit ESC that supports BLHeliS and flash BlueJay onto it.