bitdump/BLHeli #648: Tekko32 F4 45A and SPort Telemetry Frsky

sybu0, 2022-09-08:
Dear all,

I recently noticed when updating software of an old Tekko32 35A Holibro that a new option (at least for me) SPort was available and I was very pleased to see that I could connect the TX of my ESC to the SPort of my X8R receiver and get full ESC Telemetry.
It is very interesting to have full telemetry for a very light glider without FC and without a heavy Frsky ESC.

So, I bought 4 Tekko32 F4 45A to Aliexpress, but was very desappointed to see that the SPort choice is not available for this more recent ESC.  Do somebody has some information about this issue? Is there any chance to see SPort option appear for future software versions?
Kind regards

sskaug, 2022-09-08:
This is a limitation of the MCU hardware. The recently popular F4 chips have a UART that is not as flexible as the STM one, and does not have the required functionality required to implement S.PORT. Unfortunately.

sybu0, 2022-09-08:
Thank you very much for your explanations,  best regards.

sybu0, 2022-09-09:
Sorry to come back to you, I know that the connection of a F4 FC tx to the SPort of a Frsky Receiver require a signal inverter and it is not a big issue. Is this the functionality required to allow a connection with a F4 ESC? Is it possible to develop the SPort interface this way? 
Best regards

sskaug, 2022-09-09:
I don't quite recall all the details, but what generally is required is both the ability to invert, as well as the ability to do both RX and TX on the same pin

sybu0, 2022-09-09:
Well, I believe ESC is only sending data... don't you think so?

sskaug, 2022-09-09:
Regular BLHeli_32/KISS telemetry is unidirectional (TX only).
S.PORT is bidirectional.

jasc76, 2023-09-10:
WTF... I just bought one to put in a small glider to get telemetry and then such a bummer :-( 
why why why 
I had Tekkos before and they had S.Port and now that

Jetrell, 2023-12-07:
@sskaug @sybu0  Sorry to hijack the thread.. But is this also the reason **this** particular ESC doesn't support temperature via ESC telemetry ?
I had read somewhere that the AT F4 doesn't have an internal resistor to read the temperature from, as the STM F4 does.
If this is so, can you guys find a software workaround for this ?  Are there any other components within the AT processor. That could be potentially used to get an approximate temperature readout from ?

We have paid for the hardware and the software. Without there being any word of this feature being retracted by hardware limitations.. I would appreciate some form of software fix.  Because its impossible to have a fix with an external thermistor on current hardware.

sskaug, 2023-12-07:
Please check out the newly released Rev32.10 code

sybu0, 2023-12-07:
Hello, I suppose this ESC is able to handle bidirectionnal DShot, so, it means that the PWM input is bidirectionnal.
Should it be possible to use FPort protocol (SBUS & SPort on the same wiring) to make possible Telemetry connection to a FrSky ESC?
Best regards

sybu0, 2023-12-07:
Sorry... to a Frsky .... receiver...

sskaug, 2023-12-07:
@sybu0 Most likely yes. But - FPort/SPort/SBUS etc users are a very very small portion of the total BLHeli_32 user base. So it's a matter of priorities (and actually for some MCUs flash size). And at the moment we do not have any such plans.

sybu0, 2024-10-13:
For those who are still interessed by this subject, I can tell you that the telemetry pad of the Holybro Tekko32 F4 is bidirectional and if you use a bidirectional S.Port inverter, you can have full telemetry with this ESC. You have to flash your ESC to ESCape32, which is not so difficult.
[image]