bitdump/BLHeli #650: BOOT_START (BB1 & BB2) vs. BB51

stylesuxx, 2022-09-12:
I have a question regarding BOOT_START addresses: It seems on BB2, although more flash, you kept the Bootloader at the same address as with BB1 - 0x1C00 (last page of 8k flash).

On BB51 you decided to use the actual Bootloader page at 0xF000.

I was wondering if you could explain this design decision?
I mean, it sure makes sense to have "more flash available" on BB51, but why did you not decide to do something similar when initially adding BB2 support? The only reason that comes to mind is to simply keep compatibility with what was already there. I was just wondering if I am missing something...

sskaug, 2022-09-12:
Check up the flash sectioning of BB51. Larger blocks etc...

stylesuxx, 2022-09-12:
Thank you.

BB51 has bigger page size, so I get that part.
Just leaves me wondering about BB1 vs BB2, since they have the same page size but BB2 has more flash, I'll just go with "keeping the compatibility" ;-)

closed 2022-09-12 by stylesuxx