bitdump/BLHeli #651: BB51 Wrong page being erased

stylesuxx, 2022-09-12:
Sorry for being a pain in the ass - but I have one more question:

I am making adaptations on esc-configurator.com in order to being able to handle BB51 MCUs and when writing the "EEprom" section I first erase the page and the write the settings.

EEprom offset is at 0x3000 with the BB51, so page 6. I send a request for page 6 to be erased, but instead page 1 (@0x0800) is being erased.

I am sending CMD 0x39 with parameter 0x6 and receive a 0x00 ack with address 0xC00 - which would indicate that in fact the page size is ignored in the bootloader (512x6 => 0xC00).

It seems that I can mitigate this by simply multiplying the page by 4 to stay in the 512 Byte page size. I am just wondering if this is the way to go, or if it should be accounted for in the bootloader?

4712, 2022-09-13:
@stylesuxx yes, because the 4way-interface code in FC code handles the translation fixed

> // Address =Page * 512
  ioMem.D_FLASH_ADDR_H = (Dummy.bytes[0] << 1);

you should calc the page number = address / 512 -> 12288/512 -> 24

stylesuxx, 2022-09-13:
I see, that's where the translation happens.
Yeah, makes sense, then the configurator will have to adjust for that accordingly.

Thanks.

closed 2022-09-13 by stylesuxx