bitdump/BLHeli #654: When might "Dshot extended telemetry" be added to main release of BLheli-32?

jimkazmer, 2022-10-07:
I am really excited about "Dshot extended telemetry".  I ordered some new Tekko32 F4 45A ESCs and will be testing it in the next few days on a test bench setup (using the test firmware you created for this--thank you).  I am using the specification from https://github.com/betaflight/betaflight/pull/11694#issuecomment-1167992943 Is that the correct specification to use?  
Enable with:
DIGITAL_CMD_EXTENDED_TELEMETRY_ENABLE - (13) - Need 6x
and disable with:
DIGITAL_CMD_EXTENDED_TELEMETRY_DISABLE - (14) - Need 6x
?

When might this become part of the standard firmware release?

jimkazmer, 2022-10-07:
Follow-up questions...
What "States" and "Events" were implemented (if any)?
Was any other info (other than Temp, Voltage, current) implemented?  If so, what "eee0" value represents each?

haslinghuis, 2022-10-07:
@jimkazmer you might want to check https://github.com/betaflight/betaflight/pull/11694

sskaug, 2022-10-08:
Yes, the spec you link to is the one BLHeli_32 has implemented.
No states or events are implemented, only temp, voltage, current data.
We have not yet decided content or schedule for a new full BLHeli_32 release.

jimkazmer, 2022-10-08:
Thank you.
I will post my results some time next week (ESCs should arrive Monday).

sskaug, 2022-10-12:
This is actually as expected. The chip internal temperature sensor of ArteryTek F4 MCUs is not usable. Therefore we recommend manufacturers to add an external NTC temperature sensor. But not all manufacturers have done that on all their products.

jimkazmer, 2022-10-12:
Hmm, and I think I read another post you answered that the current can't be reported on the F4 MCU... yet if they claim "current-sensing" and the BL-Heli configuration shows the config element to limit current, why can't current be reported via telemetry?

sskaug, 2022-10-12:
Sometimes the current sensing element is only one per 4in1, and then it is most often just analog output to the FC. Although some also feed this to the ESC for current sensing (but not protection). 
Of course it can also happen that manufacturers use code that supports current limiting, without having hardware for it...

jimkazmer, 2022-10-12:
Is there a way to test if the manufacturers "current sensing" claims are valid?  They are claiming the single Tekko F4 45A ESC has current sensing.

jimkazmer, 2022-10-12:
I would love to know which (single) ESCs you (and your peers) think are great ESCs.

sskaug, 2022-10-13:
If you see a single ESC with a current sense resistor, like the Tekko32 F4, then it will have current sensing. AFAIK, this is a good ESC.

jimkazmer, 2022-10-13:
Thank you for the replies and help.  I am waiting for a response from Holybro regarding the "missing" current telemetry value. I'll add whatever I learn from them here, then close this issue.

haslinghuis, 2022-10-18:
@sskaug it would be nice to have a list with builds and there options supported. Maybe have certification for a select group of manufacturers / boards.

sskaug, 2022-10-19:
That is a long list.... We did start that regarding telemetry port, it's in the manual. But has not been updated lately due to the list getting very long and lack of feedback that it was really useful. There is also the problem of mapping code name to actual product. We only know the code names. And buyers generally only know the product names.

jimkazmer, 2022-10-26:
An update... sort of...
HolyBro has not been able to explain the lack of the current value for telemetry in their Tekko32 F4 45A ESCs.

**Worse however is a behavior that when the ESC encounters a desync, it stops working until I power cycle.**

This makes these ESCs un-usable for me.  This sounds like the same problem they were having with their F3 based ESCs.  Uhg... not sure what to do about this.

jimkazmer, 2022-10-27:
Update to above...
I changed "Stall Protection" to "Relaxed" and **the problem still occurs**... it may have taken longer, but the ESC/motor stopped and a power cycle was required to restore function.

I changed "Stall Protection" (back) to "Normal" and then changed "Demag Comp" to "Low" and **the problem still occurs**... it happened quicker than the above test, but the ESC/motor stopped and a power cycle was required to restore function.

I changed "Demag Comp" to "Off" (Stall Protection=Normal) and **the problem still occurs**... it happened just as quickly as the above test, but the ESC/motor stopped and a power cycle was required to restore function.

I changed "Demag Comp=Off"  and "Stall Protection=Relaxed" and **the problem still occurs**... it took a bit more time, but the ESC/motor stopped and a power cycle was required to restore function.

Once again, the other ESCs on the test bench never "deactivated" and required a power cycle to restore functionality. They are all older models (which I cannot buy any longer... hence the quest for finding their replacements).

jimkazmer, 2022-10-27:
Another update...
I also tried playing with "Low RPM Power protect" equal to "Adaptive" and "Off".  The problem (motor deactivates until a power cycle) still occurred at 4S for both those settings.  When "Low RPM Power protect=Off" the problem did not occur at 3S (but maybe I didn't restrain it for long enough).

jimkazmer, 2022-11-03:
Another update...
The motor being stalled, triggering an ESC/motor "deactivation", must be common behavior for some ESCs.  I bought a different ESC with current sensing that behaves the same way. My older ESCs, without current sensing, never "deactivate".

Can someone explain this behavior?
Is there a way to configure the ESCs so the ESC/motor does not deactivate?

closed 2022-11-04 by jimkazmer