bitdump/BLHeli #655: Does "motor coasting" (i.e. Brake on stop = Off) work the same for forward and reverse powers for "Bidirectional 3D"?

jimkazmer, 2022-10-10:
The motors, when powered in the forward direction, take a little bit of time to spin down to stopped (coasting to a stop). 
However, when then motors are powered in the reverse direction, they stop extremely quickly.

Is there a difference in how "Brake on stop = Off" behaves for forward and reverse with "Motor Direction = Bidirectional 3D"?

Below is an image of my ESC settings.
[image]

sskaug, 2022-10-13:
When doing a quick test here, the motor stops quickly in both directions. And this is as it shall be, as it is not nondamped mode and hence active braking is engaged. After motor is "declared" stopped by firmware (at some low rpm), it is not driven anymore and can spin a bit freely before stopping.

jimkazmer, 2022-10-13:
Thank you for checking.  Based on the behavior I am seeing I am thinking the braking rates may not be equal (between forward and reverse directions), but I will do some more extensive testing and measurement when I have time.  I am closing this ticket.
Thanks.

closed 2022-10-13 by jimkazmer