bitdump/BLHeli #658: Variable torque at low to moderate RPM

ChrisRosser, 2022-10-30:
Following some testing of BLHeli32 with an inertial dyno I noticed variable torque at moderate RPMs.

It is present regardless of PWM or timing setting and has been confirmed to affect Mamba F55 Pro 128K and Tekko32. This issue is not present with AM32 software which makes unlikely to be a test system issue.

Mamba F55 Pro
[image]
[image]
Tekko32
[image]

sskaug, 2022-10-31:
Thank you for sharing, this is interesting data. Some questions to best understand the data:
Being an inertial dyno, I assume each trace is taken during one acceleration run?
What is the time it takes to do that run?
Is the RPM data mechanical RPM? Do you use a 14pole motor?
Do you know why BL32  torque starts at a lower rpm than AM32?

ChrisRosser, 2022-10-31:
Yes each trace is taken during one acceleration run which takes approximately 2 seconds. Logging is at ~100Hz.
RPM and torque are both mechanically measured: RPM using an optical tape, torque using 2 load cells.
The motor is a XING2 2207 1855KV with 14 poles.
I'm unsure why the starting RPM and Torque are different for BL32 vs AM32. The test is as follows.

1100 throttle for 5 seconds.
Start logging.
1600 throttle until 25000 RPM.

sskaug, 2022-11-01:
The difference at 1100 is most likely due to different settings for minimum throttle.
I do have a hypothesis for the torque ripple you see with BL32.
I assume you use a standard servo tester? In that case, throttle signals are most often sent at a 50Hz rate, meaning that there is a new throttle pulse every 20ms.
BL32 limits max power for low rpms (as I'm sure AM32 does too), and this power limit is updated on every input pulse. It is possible that this is what causes the ripple you see. In a "normal" multirotor environment this is not an issue, since there the input signal rate is high enough to be smoothed out by the low bandwidth of the mechanical system.
In your testing, can you try to increase rampup power?
And/or disable low rpm power protection?

ChrisRosser, 2022-11-02:
Hi,

Here is a test at Dshot600

[image]

We see a big spike in torque followed by a plateau and then a gradual decline.

cbf123, 2022-11-04:
It might be useful to show the current draw on the same graph.  In the Youtube video comments you mentioned that the current consumption drops relatively smoothly while the mechanical torque varies.

ChrisRosser, 2022-11-05:
Sure @cbf123 . Here it is!

[image]

sskaug, 2022-11-05:
Thanks for testing again. I assume your Y-axis is still torque, and that the title "Motor Optical Speed (RPM)" is misleading?

Now we do not have a test setup similar to yours, we generally test with the load that the ESC normally sees (i.e. prop :)).
Replicating a test like yours, where throttle is increased over about 2 sec (gradually, as immediate acceleration of course will be very fast), this is what we get:
[image]
[image]
Some ripple in the acceleration curve, but not very significant.

Now, there are some features in BLHeli_32 (which are similar to those in BLHeli_S) that can be potential causes for ripple. The most likely is the extension to the digital bemf filter designed to reduce effects around where motor pwm frequency matches commutation harmonics. Another might be variable averaging weight of commutation period updates.
With those two modified, this is what I get:
[image]
Maybe less ripple, or maybe just differences in measurement noise.

Anyway, if you can test these modifications, please let us know the exact code you use (the name that shows up in BLHeliSuite32 or the Android app).

ChrisRosser, 2022-11-05:
@sskaug Happy to test. MAMBA F55 128K 4110

I have to mention that a slow throttle ramp is a very different test from an inertial dyno run. Ramping throttle slowly requires minimal torque from the motor whereas the dyno demands much more torque at the same RPM.

sskaug, 2022-11-05:
Sure, such a dyno test condition probably resembles e.g. car operation conditions more than air operation conditions.
Great that you can test. Here are three testcodes: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes
The "MC" code has the bemf filter feature disabled. The "FC" code has the variable commutation period averaging weight feature disabled. And the "BOTH" code has both disabled.

ChrisRosser, 2022-11-07:
Test results:
MC Firmware
[image]
FC Firmware
[image]

I think we have our culprit.

sskaug, 2022-11-07:
The above plots are more informative than the initial plots you shared, as individual measurement points are now shown.
What is it that determines the sample rate in your system?
Are the above curves taken with a high rate input signal (like e.g. Dshot600 at 4k or more)?

sskaug, 2022-11-07:
The low rpm power protection scheme use has a lot of inheritance from older BLH versions, and looking at is it in detail, the quantization steps at high power levels can be noticeable, particularly for the high inertial load system. So a new testcode is published on github "LOQ", reducing these quantization steps. Would be great if you could test :).

ChrisRosser, 2022-11-07:
> Great to hear, thank you for testing. The effect from the bemf filter feature is noticeable but probably not very significant in most systems. And with "byRPM" pwm, it will not be necessary. I think we can consider removing it and use a code similar the the "MC" testcode, going forward.

I would agree. For multirotor applications torque required is minimal at low RPMs most of the time. Slightly better responsiveness might be possible with more torque but at very fast change of RPM a lot more comes into play. My test system is not yet capable of measuring such fast phenomenon.

ChrisRosser, 2022-11-07:
> The above plots are more informative than the initial plots you shared, as individual measurement points are now shown. What is it that determines the sample rate in your system? Are the above curves taken with a high rate input signal (like e.g. Dshot600 at 4k or more)?

The max sample rate I can do is 100Hz. How that relates to RPM points depends on how fast the system accelerates. I can use heavier flywheels to slow things down but that can increase the peak current and increase the risk of smoking motors.

ChrisRosser, 2022-11-07:
> The low rpm power protection scheme use has a lot of inheritance from older BLH versions, and looking at is it in detail, the quantization steps at high power levels can be noticeable, particularly [...]

Yep happy to test!

ChrisRosser, 2022-11-08:
[image]
LOQ test code doesn't seem to have made much difference.

sskaug, 2022-11-10:
Thanks again for testing. So there appears to be kind of a plateau from 5-7000rpm.
At this point the only rpm dependent mechanisms in the code are bemf sample averaging and low rpm power protection.
Still can't figure out  how any of them should give any plateauing...

ChrisRosser, 2022-11-12:
> Thanks again for testing. So there appears to be kind of a plateau from 5-7000rpm. At this point the only rpm dependent mechanisms in the code are bemf sample averaging and low rpm power protection. Still can't figure out how any of them should give any plateauing...

Could it be something time based? Is there a certain throttle setting for a number of ms on a sudden increase? Could it take a few ms to update a certain value/parameter to reflect the new higher throttle setting?

MartinWa, 2022-11-12:
> Could it be something time based? Is there a certain throttle setting for a number of ms on a sudden increase? Could it take a few ms to update a certain value/parameter to reflect the new higher throttle setting?

A slower ramp up would show that I guess? Perhaps testing that is hard to do, I guess it would need a larger wheel and that might smoke the motor?

ChrisRosser, 2022-11-13:
@MartinWa I can try a ramp up with a larger flywheel. I can also do ramps at various throttle settings. Currently I've left rampup power at default. Here are the options:

2 wheels 100/200 kgmm2
Throttle settings 20-50%
Rampup power

Let me know what you want to try

sskaug, 2022-11-13:
> Could it be something time based? Is there a certain throttle setting for a number of ms on a sudden increase? Could it take a few ms to update a certain value/parameter to reflect the new higher [...]

Except for limiting mechanisms (low rpm, voltage, current, temp, spoolup) there are no limitations on throttle. Every input dshot frame is immediately converted to and applied as the corresponding motor pwm.
Presumably in the testing, voltage, current and temp limiting does not apply. And the motor is already spooled up.
Low rpm limiting limits power linearly from about 32000 erpm (default rampup setting) and down.

slackista, 2023-02-23:
Are there any research going on with this issue?

kenkamm, 2023-06-11:
Has this issue been resolved, and if so, in which version?

sskaug, 2023-06-11:
Improvements are implemented in the testcode here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Dshot%20extended%20telemetry%20testcode

kenkamm, 2023-06-11:
Thank you! Will this new code make its way into the official release? If so, when?

sskaug, 2023-06-11:
It will, but not yet known when. We would like to see many installs of testcode before we publish an official release.

truglodite, 2023-07-25:
Curious how you count the number of installs of testcode?

FWIW, I'm flying 32.95 testcode on a tekko32 f4 mini 45 v1.3, and it's working great.

sskaug, 2023-07-26:
Great to hear that it runs well for you. We count installs when BLHeliSuite32 or the BLHeli_32 app validates the serial number upon code flashing. 32.95 installs are increasing, not very fast, now in the few hundreds, it seems generally people are happy also with 32.9.