bitdump/BLHeli #659: Missing or unsuitable OpenSSL library on MacOS Ventura

jgalerap, 2022-10-30:
Got this error message using version 1041 for Mac after upgrading to MacOs Ventura.

[image]

OpenSSL is properly installed. 

aaa@none ~ % /usr/bin/openssl version -a
LibreSSL 3.3.6
built on: date not available
platform: information not available
options:  bn(64,64) rc4(16x,int) des(idx,cisc,16,int) blowfish(idx) 
compiler: information not available
OPENSSLDIR: "/private/etc/ssl"

4712, 2022-10-30:
Please use version 1.0.4.2 form [here](https://drive.google.com/drive/folders/1Y1bUMnRRolmMD_lezL0FYd3aMBrNzCig) or [here](https://www.mediafire.com/folder/dx6kfaasyo24l/BLHeliSuite)

jgalerap, 2022-10-31:
That worked. Thank you!

closed 2022-10-31 by jgalerap