bitdump/BLHeli #662: ESC deactivates after stall condition

jimkazmer, 2022-11-04:
I have two current sensing ESCs that both deactivate the ESC/Motor when a persistent stall condition occurs:
(a) Tekko32 F4 45A (TEKKO32_F4_Multi_32_91.Hex)
(b) RDQ 3-6S 32Bit DShot1200 30A (RDQ_32_Bit_AT_Multi_32_91.Hex)
I have an older ESC, without current sensing, that does **not** exhibit this behavior:
(c) SkyStar 50A (KS_BLHeli_32_50A_V1_Multi_32_91.Hex)

Once the stall condition is removed (my hand lets go), the motor will not turn (it sometimes creates a squeal with no movement). Only a power cycle restores function.

Can someone explain this behavior?  Is it expected?
Is there a way to configure the ESCs so the ESC/motor does not deactivate?

Below are my initial settings, but I have tried many other combinations of settings (listed below picture).  It is easily reproduced on both 3S and 4S. A prolonged stall causes the ESC/Motor to "hang".
[image]

I changed "Stall Protection" to "Relaxed" and the problem still occurs... it may have taken longer, but the ESC/motor stopped and a power cycle was required to restore function.

I changed "Stall Protection" (back) to "Normal" and then changed "Demag Comp" to "Low" and the problem still occurs... it happened quicker than the above test, but the ESC/motor stopped and a power cycle was required to restore function.

I changed "Demag Comp" to "Off" (Stall Protection=Normal) and the problem still occurs... it happened just as quickly as the above test, but the ESC/motor stopped and a power cycle was required to restore function.

I changed "Demag Comp=Off" and "Stall Protection=Relaxed" and the problem still occurs... it took a bit more time, but the ESC/motor stopped and a power cycle was required to restore function.

I also tried playing with "Low RPM Power protect" equal to "Adaptive" and "Off". The problem (motor deactivates until a power cycle) still occurred at 4S and 3S for both those settings.

sskaug, 2022-11-04:
Thank you for reporting. This will need to be further investigated. There is no mechanism in firmware that deactivates the ESC until cycling power.

jimkazmer, 2022-11-05:
It isn't a bad feature... but I'd need a dShot command to indicate a reset (to avoid the power cycle), and maybe a status indicator that the ESC/Motor has triggered the condition.

jimkazmer, 2022-11-07:
I tried turning OFF current protection... same behavior (stall condition deactivates ESC/motor).
I changed the firmware back to 32.9 (from the 32.91 beta)... same behavior (stall condition deactivates ESC/motor).

jimkazmer, 2022-11-07:
I tried the settings shown below, and it happened quicker than ever, with the least force required to stall the motor.
[image]

jimkazmer, 2022-11-07:
I have a competition this weekend, and need to use these ESCs (I do not have enough of my old ones).  Any insight about how to avoid the issue (via config settings) would be appreciated.

I tried the settings below (setting by RPM to 16K both high and low), and the problem still occurred.
[image]

jimkazmer, 2022-11-07:
Here is a video showing the problem: https://youtu.be/aultj1uf7qI

I changed my ESC settings to the ones that made it the hardest to reproduce (shown below).  The ESC/motor on the left are my old SkyStar 50A ESC, on which I could never reproduce this issue. The ESC (on the right) being used for this video is shown in the screen shot below.

[image]

sskaug, 2022-11-08:
Sorry, I wasn't paying full attention to your setup. In bidirectional 3D mode, it can happen that the system doesn't exit run mode even if throttle is zeroed. And the squealing noise that you hear is the code actually believing that the motor is running, due to feedback from coil currents to bemf sensing comparator.
Have you tested the "loaded startup testcode"?: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Loaded%20startup%20testcode
It is designed to be more resilient to such cases.

jimkazmer, 2022-11-08:
The "loaded startup testcode" does not have firmware files that match the RDQ and TEKKO firmware files I am using:
- RDQ_32_Bit_AT_Multi_32_91.Hex
- TEKKO32_F4_Multi_32_91.Hex

I'm not sure what to use...

sskaug, 2022-11-08:
I see, your are new and it's been some time since I generated the loaded codes.
I've put two "PLANENOWINDUP" codes for you here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

jimkazmer, 2022-11-09:
That code did stop the "ESC/Motor deactivation" (on both the RDQ and TEKKO ESCs). Thank you!  I heard the squeal, once out of many tries, while I was still holding the motor, but the motor resumed it function once I let go.

It seems that motor spin-up (acceleration) is much much slower. Is that expected?

Does that new test code also include "dShot extended telemetry"?
Once again, thank you.

sskaug, 2022-11-09:
Great to hear that it works as intended!
I should publish a new version of this modification based upon 32.9 someday soon.
Yes, the code has the extended telemetry.
And yes, the slower acceleration comes with the modificaiton.

jimkazmer, 2022-11-09:
Does the slower acceleration persist at higher motor powers (e.g. 40%, 60%, 80%)?

Does "Rampup Power" or "Maximum Acceleration" config settings (or any other setting) affect the rate of acceleration?

The rate of speed changes is of utmost importance in my application.

Is there a thread somewhere that I can read more about this?

fkoran, 2023-03-05:
I encountered the same "squeal direction reversal" issue with the Holybro Tekko32 Mini F4 45a ESC and T-Motor 1507 motors. Flashing with TEKKO32_F4_45A_4in1_Multi_32_99.Hex resolved the issue