bitdump/BLHeli #667: What are X and Y Layout "reserved" for

stylesuxx, 2022-11-29:
A manufacturer stepped up to us (Bluejay) with a new layout and we were wondering what [X and Y](https://github.com/bitdump/BLHeli/blob/4df6e6fbaaa5a0ef36fa20c4628c5538a67a6abe/BLHeli_S%20SiLabs/BLHeli_S.asm#L131) layouts are reserved for, or if they can be used? If not - where there any plans of extending the naming scheme and if so, how?

sskaug, 2022-11-29:
These layouts were reserved a long time ago. And still not used. So I do not think we should reserve them anymore. I will inform the affected parties.

sskaug, 2022-11-29:
Now sent info emails to the two parties. Hopefully there are no heavy arguments against freeing up X and Y.
As for extension beyond Z, we haven't thought about that. Really the existing layouts should provide sufficient flexibility for most compromises with hardware.
Also - hopefully there will now be a shift to BB51. It is a full function chip that is cheaper. And with the chip shortage now hopefully soon ending, it should also become readily available.

stylesuxx, 2022-11-29:
I understand. Thank you - keep me in the loop.

In this case it seems to have been negligence from the designer as they swapped PWM and COM pins around, but we are still trying to find out if this is the only issue. So probably this layout would be limited to a very small run and be fixed in the next revision. I still think there should be an "official" layout for that, otherwise users might end up having issues...

sskaug, 2022-11-29:
I see. With only X and Y still free, is it right to assign one of them to a run of only some few units?

stylesuxx, 2022-11-29:
I will speak to the manufacturer and see how many units this actually is. What would be your suggestion? I am afraid that otherwise this will lead to problems in the future. If it's just a couple units, then they could just use those as review samples or so and inform those users... - but if this goes out to a bigger user base, sacrificing one layout would probably be the right thing to do.

sskaug, 2022-11-29:
I'd say it should rather be thousands...

sskaug, 2022-11-30:
For the X layout, we now have received acknowledgement for that it can be reassigned.

stylesuxx, 2022-11-30:
Awesome, thank you - still waiting on an absolute amount from the manufacturer...

stylesuxx, 2022-12-01:
Alright, manufacturer replied back and will do the responsible thing. They will only use the "wrong" layout internally for testing and will not ship them to users.

So sorry for bothering you with this, but at least we now have 1 spare layout confirmed ;-)

closed 2022-12-01 by stylesuxx