bitdump/BLHeli #668: Holybro Tekko32 F4 4in1 65A reboots at PWM >72khz

blind-oracle, 2022-12-04:
ESC: Holybro Tekko32 F4 4-in-1 65A [url](http://www.holybro.com/product/tekko32-f4-metal-4in1-esc-65a/)
BLHeli32: 32.9

Problem is simple:
if I set PWM (constant, with variable it's harder to test as it's throttle or RPM dependent) to something higher than ~72khz (proven to be safe, but maybe 74-76 would also work) - the ESCs just reboots after some minutes of flying.

It's not a desync - quad just drops and I hear the startup melody. So I guess all 4 CPUs reboot at the same time since all props are stopped exactly at the same time.

Any ideas what can cause this? It was the same with 32.8 and maybe earlier.

sskaug, 2022-12-05:
This sounds like it may be hardware related, you should probably contact your vendor.

blind-oracle, 2022-12-07:
@sskaug thanks. Well, I contacted the vendor and they just relayed the problem to Holybro. Since my ESC is out of warranty probably that would yield zero results... 🙄

What is strange that it works perfectly on lower frequencies.