bitdump/BLHeli #669: Motor stop spinning after hard crash, power cycle required

nicesub, 2022-12-06:
Hi,

I have Mamba stack FC MK4 APP with 55A ESC. What I have noticed that on some crashes which impact the motors the motor which potentially got hit in the crash does not spin again after the crash. I have to power cycle the drone to sort out the issue. It's never the same motor affected which leads me to believe that it stops turning the motor which maybe got physically stopped on the crash. 
I didn't mess with the configurator too much except of increasing PWM frequency and putting on custom startup tune. Current firmware on it is latest public available blheli32. 
Apart from this everything works fine.

jimkazmer, 2022-12-06:
I think this is the same issue as #662

nicesub, 2022-12-06:
Read thru 662, not sure I understood what is the solution?
Also on my end, the drone is not making any strange noise after the motor stops spinning.

sskaug, 2023-05-26:
@ewileycoy Thank you for putting focus on this again. The 32.99 code is not designed for use in planes or copters, it was created for use with underwater thrusters. 
I'll be out for this weekend, but if would be willing to help test modified codes, we could work towards improvements next week. 
If so then we will need the exact name of the code you are using (the name that shows up in BLHeliSuite32 or the BLHeli_32 Android app).

ewileycoy, 2023-05-26:
Thanks for the response, i'm using "SKYSTARS_KM60_Butter_128k" Rev. 32.9

sskaug, 2023-05-30:
Thank you for reporting in detail, I believe we have now reproduced the issue and created a fix for it.
Testcode (Rev32.94) for your ESC is posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

ewileycoy, 2023-05-31:
Awesome, thanks so much. I loaded the new firmware and it seems to be working after a few test crashes, though i'll test more thoroughly this weekend.

sskaug, 2023-06-07:
Rev 32.94 code is now published for all targets

CruiseFPV, 2023-06-09:
I am having this same issue, how an i flash this pre-released Firmware please? I've tried "save as" of my esc file to the BLHeli32_HexFiles folder, then opened blheli configurator, flash blheli, but it only shows 32.8 & 32.9, i then tried clicking on ignore and went to the HexFiles Folder to manually flash, but the hex files would not show up?