bitdump/BLHeli #671: Temperature control and telemetry bug on BB21 1S builds

damosvil, 2023-01-02:
Temperature telemetry reported on 1S is lower on BB21 1S builds. Seems that the root of the problem might be that on 1S there is no external voltage regulator, so as voltage on 1S is fluctuating ADC has no stable VREF to perform conversions:
[image]

It also affects to temperature control on 1S builds.
This bug has been found in Bluejay, that uses the same ADC config than Blheli_S to read temperature.
Please, do you know of any method to use a stable ADC VREF, so temperature can reliably be measured at 1S?
https://github.com/bird-sanctuary/bluejay/issues/53

sskaug, 2023-01-02:
Using 1.65V or 2.4V reference as you propose in [bird-sanctuary/bluejay#53](https://github.com/bird-sanctuary/bluejay/issues/53) is probably the way to go for Bluejay. BLHeli_S is no longer being developed.

damosvil, 2023-01-02:
Ok, we will try that way. Many thanks