bitdump/BLHeli #673: 32.7 Firmware request for T-Motor Velox 45A V2

StuartD89, 2023-01-09:
Hello, is it possible to get a 32.7 firmware target for the T-Motor Velox 45A V2 ESC please, I can only see 32.9 in the configurator.

The website says the firmware target is : "Tmotor_32Bit"

But when I go to flash it says "Tmotor_FG0_01 Rev 32.9"

[image]

This is the ESC link: https://www.hobbyrc.co.uk/t-motor-velox-v45a-v2-4in1-esc-3-6s

Thanks

sskaug, 2023-01-09:
This code was generated after Rev32.7 was released. Why do you want 32.7 code for it?

StuartD89, 2023-01-09:
> This code was generated after Rev32.7 was not the latest release. Why do you want 32.7 code for it?

The main reason is that I don't like how 32.9 flies, I much prefer 32.7, but I also get minor desyncs when I run 32.9 on my main quads using T-Motor F55A Pro 2 ESC's. Is it possible to get a hex for 32.7 on this ESC?

Thanks

StuartD89, 2023-01-09:
> Now posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

Thank you so much!

StuartD89, 2023-01-09:
> Now posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

I'm getting this error when I try to flash

[image]

sskaug, 2023-01-09:
You must click "raw" and then do save as:
[image]

StuartD89, 2023-01-09:
> You must click "Raw" and then do save as: [image]

Lovely, thats working now, thank you so much for your help!