bitdump/BLHeli #674: Loaded startup code for Readytosky 35A AT32

smelting, 2023-01-12:
I'm having trouble with a newer version of the Readytosky 35A single ESCs with an AT32 micro. Previously I've had no issues running in the bidirectional mode for use with ground vehicles. My setup is an 1806 connected to a gearbox directly driving a wheel. Older STM32 versions of this ESC did not show this issue. I've used the exact same parameters as the STM32 variants. This issue looks similar to #662, I've linked a video showing the behavior. The issue can be replicated by rapidly changing direction from full speed one way to the other. Slow inputs don't always seem to cause the issue. Normal stick input while driving around will cause the issue.

[image]

https://drive.google.com/file/d/1XMvR76VkRelSVxeK3WaROi_XbpEHwTpb/view?usp=sharing
Power cycling the ESC will fix the issue. The issue does sometimes clear if you manually rotate the motor (by spinning the wheel in this case). The gearbox is 20:1 so when you manually turn the wheel the motor is spinning decently fast. 

I've tested this with multiple ESCs and it happens on all of them.

sskaug, 2023-01-14:
First of all, this mode of operation is demanding. What most likely happens in your case, is that there is feedback from motor currents to the bemf sensing comparator. So due to this feedback, the code believes the motor is running, while actually it is not.
There are many components in this picture, the quality of the comparator - input offset voltage, input common mode voltage etc. Then there is the accuracy of the bemf sense dividing resistors. And of course pcb effects, like coupling from trace to trace, noise on the ground plane etc.
I do not have your specific hardware here. But I have another ESC with the same MCU (AT32F415), and it behaves very well on my bench here. Maybe you will have to try another ESC make for this application.

smelting, 2023-01-14:
Could you generate some loaded startup test code for my specific target to try out? I have a large number of these ESCs and would like to be able to use them.

sskaug, 2023-01-15:
Now posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Loaded%20startup%20testcode

smelting, 2023-01-15:
Thanks for posting the test code, I tried it out with no luck (same settings as posted above). Below 5-10% throttle the motor runs fine. Once you go above that it seems to lose sync of the motor and starts freaking out. The motor I had connected started smoking after 3 seconds or so at 50% throttle. I've linked a video to show the behavior. https://drive.google.com/file/d/1bTfGh-mEoL0xI8KJXtjmLe5RaNHbPy2r/view?usp=share_link

Any ideas on what might be causing this? As I said I have a large number of this hardware variant and need to get them working in bidirectional mode.

smelting, 2023-01-17:
Thanks, 32.9.9 worked well. What ended changing under the hood? Do you think loaded startup will ever become a parameter in the configuration suite? Or maybe more options for helping down at the low end of rpm?

sskaug, 2023-01-17:
Great that it now works! The change was to the filtering of the bemf signal. There are surprisingly few installs of this code, so at the moment we can not see that one or more configuration parameters are justified.

VTFPV, 2024-04-25:
Hello @sskaug ,  I am having a similar issue with a Hobbywing_BL32_AT421.  Motor/motors stalling and not restarting after a crash.  Re-cycling the power brings everything back to normal again.  I cannot find a hex file for the Hobbywing_BL32_AT421 in the link above (https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Loaded%20startup%20testcode) can I use another hex file or could you add a hex file for this ESC please.

sskaug, 2024-04-25:
Sure, HW codes are now added

VTFPV, 2024-04-25:
Thank you @sskaug  😊